sebanc/brunch #607: Sound not working

gjbnh1701, 2020-10-27:
I continue to be BLOWN away at how good this is..But I upgraded my AIO in the last week and sound no longer works. It has an Intel/Realtek HD chip-set in it...works out in the box under linux...is there a way to pass a command to the kernel?? Otherwise it is flawless!!

sebanc, 2020-10-28:
I am not sure I understand, did sound work on previous brunch versions ?

Which brunch / chromeos versions are you on ?

gjbnh1701, 2020-10-28:
chrome os 85 and Brunch 85...used the same usb key on previous system, no problems...not the case on new system....even rie did chromeos on new system under windows AND linux. The wireless changed which I fixed, so I can get online. But no sound now. Windows identifies sound as Realtek High Definition Audio.

sebanc, 2020-10-28:
Ok so sound never worked with brunch on the new system if I understand correctly. What's your computer model ? Also, could you try to update to chromeos r86 by enabling updates ? (There was a bug with audio in r85)

gjbnh1701, 2020-10-28:
HP AIO Ff0003w is the model...

sebanc, 2020-10-28:
I added you to "brunch-next" repository which has kernel 5.4 (you should have received a github invite), could you try it and see if sound works ?

gjbnh1701, 2020-10-28:
Will try ASAP...

sar777-dev, 2020-11-01:
when you are satisfied please close this issue.

closed 2020-11-03 by gjbnh1701

Ahmettrkr0, 2022-01-12:
Hello, I did a bios reset there is no sound in chrome os, but there is sound, I need urgent help

Ahmettrkr0, 2022-01-12:
@gjbnh1701 @sebanc