sebanc/brunch #608: Developer Mode and Brunch Releases

ms1000000, 2020-10-27:
Hi,
I have 2 questions:
1. I read #521 and I have a question: From which version does Brunch boot to developer mode?
2. How does Brunch releases work? Will release named r80 will work with ChromeOS 83? And what does "Note: ChromeOS r** recovery images are not yet available but this release can also be used with r** recovery images." mean?
Thanks

sebanc, 2020-10-28:
Hi,
1) brunch has always booted in developer mode by default as the access to cros shell is needed for debugging issues, grub configuration modification...
2) brunch releases are built against specific shared libraries included in chromeos recovery images. Using the same brunch and chromeos version ensures that you will not face incompatible shared libraries issues. However, in recent updates, the chromeos shared libraries have not changed much so the brunch releases are generally backwards compatible and also work with beta/dev channels.

ms1000000, 2020-10-29:
Which shared libraries? Will using older version of brunch (or just modification of the code to disable the shared libraries blocker) will affect my experience with ChromeOS?