sebanc/brunch #609: sudo bash chromeos-install.sh -src < path to the ChromeOS recovery image > -dst chromeos.img command not working provide new command works on wsl

Shiv-spec, 2020-10-28:
[image]

Lucaacer, 2020-10-28:
You don't have the utilities, you install them before

sudo apt install pv tar cgpt

Shiv-spec, 2020-10-28:
I had downloaded all the file but same error is coming

Shiv-spec, 2020-10-28:
S
[image]
ee I had reinstalled the linux distribution and installed packages same error is coming . I am uploading photo

sebanc, 2020-10-28:
It seems that is a specific issue with Kali, I have not really looked into it yet, download ubuntu WSL instead

Shiv-spec, 2020-10-28:
the same issue is coming with ubuntu wsl and linuxmint ,xubuntu iso can you please provide a command that can run without any problem or provide chrome os iso

Shiv-spec, 2020-10-28:
and thanks for comment

Shiv-spec, 2020-10-28:
or a command that can run on command prompt or powershell

daemongloom, 2020-10-30:
Can you show us lines 5-10 from your chromeos-install.sh? That error looks like you have modified this script.

Shiv-spec, 2020-11-10:
my install.sh file

Shiv-spec, 2020-11-10:
can you create iso for me because i had tried all the linux possible