sebanc/brunch #610: Autoshutdown after designated time in sleep

koehnej86, 2020-10-28:
Hello, first I would like to say how amazing you all are that started and maintained this project. I'm currently running the latest brunch on v85 nocturne from a usb drive on a Surface Pro 6. I was wondering if there is a way to automatically shutdown the pc after being in the suspended mode for a designated time. If I'm not returning to the PC before 30 minutes, I can care less if I have to go through the boot process again, the battery savings is much more worth than my time. I changed the settings in "/use/share/power_manager/shutdown_from_sleep_sec" but it has no effect. Thanks in advance!

librehat, 2020-11-09:
I'm having exactly the same issue. Everything so far works very well except resuming from sleep. I'm using the latest v86 rammus image on my Lenovo Ideapad Yoga (6th gen Intel)

librehat, 2020-11-11:
A bit more finding on this for my machine. Exactly the same brunch frameworks and chrome os recover image (rammus). When I installed it on external HDD, it'd freeze when waking from sleep, but it doesn't now I installed it onto my internal SSD (using `img` file method on README).

I suspect that sleeping also put USB devices into sleep (?) which might cause some problems for the external HDD