sebanc/brunch #612: Reboot in 10 seconds after wake up

andreabaiardi, 2020-10-28:
Hi,
i enjoy in this great project since the beginning of 2020.
I use via usb external ssd with rammus 81 and brunch 81. 
last night windows s.o. (on internal disk) that I use in the evening, it notified me some updates: among others a firmware update appeared, I proceeded with the updates, restarted windows and everything works correctly.
this morning, as always, I started the external ssd,
chrome os worked fine, then I put it to sleep with the command win + shit + l and the computer went into sleep.
when I tried to wake up the computer it asked me for the password, but after a few seconds it shut down and restarted by itself.
I tried several times, turned off, turned on again, logged in, used chrome os, then put in suspension: every time it restarts from suspension after sending the password it turns off and restarts.
I also tried to wait and not put the password, after a few seconds it restarts by itself.
while I was there, since the rammus and brunch versions were old, today I reinstalled the rammus 85 and stable brunch 86 (20201014) versions on the ssd.
unfortunately the problem remains, after the suspension, when you wake up the computer restarts.
could someone help me?
I have already checked and disabled the TPM but nothing has changed.

unfortunately I think that the firmware update has changed something in the energy management of the pc (IMHO).
Desktop Acer Veriton N4660G i3-9100T 8 Gb ram

Thanks.

Andrea

sebanc, 2020-10-28:
Hi,

Indeed it seems the firmware update changed something to usb power mangement which causes the system to reboot.

I don't have much solutions for that:
1) I added you to "brunch-next" repository with 5.4 kernel (you should have received a github invite). You can try it and see if it helps with suspend
2) If it does not work, try to identify your usb device in sysfs and to set "power/control" attribute to "on" before suspend. If it works, it can be automated with a bootscript

andreabaiardi, 2020-10-28:
Thanks Sebanc,
very quick response!
In next 2 days I'll try your solutions.
Then i'll report to you.
Thanks again.
Have a nice day.
Andrea

andreabaiardi, 2020-10-28:
Hi Sebanc,
great news...
in Italy we use to say "whoever lasts it wins" ...
after many attempts I managed to downgrade the bios firmware, now everything is back to work, including suspension and awakening.
fuck off to Acer bios firmware update...(sorry for the language but it was necessary!).
thanks in any case for kind support!
Have a great, nice, wonderful day!!!!!!!!!!
Andrea

closed 2020-10-29 by andreabaiardi