sebanc/brunch #613: HP Elite x2 G1 1012 - power button not working

meciu, 2020-10-28:
Device - Detachable notebook HP Elite x2 G1 1012
Tested builds - EVE, Rammus, Chell

Pressing power button while in CroOS does nothing (in both Notebook and Tablet mode) so I'm unable to put device to sleep without connecting detachable keyboard.

ameyjoshi3068, 2020-10-29:
Have you tried the firmware option acpi_power_button ?

meciu, 2020-10-29:
With option acpi_power_button pressing power button opens power menu with Shutdown, Sign out, Lock and Feedback (in both Notebook and Tablet mode)
Still I'm unable to put device to sleep.

ameyjoshi3068, 2020-10-31:
> Pressing power button while in CroOS does nothing (in both Notebook and Tablet mode) 

acpi_power_button can only fix this issue.

ameyjoshi3068, 2020-10-31:
As far as I know, you cannot trigger sleep manually without a keyboard
But you can select to sleep automatically through chrome://os-settings/device/power which is on by default

It would be nice feature request for chrome OS.

sar777-dev, 2020-11-04:
I found someone saying that pressing win+shift+L will put your device to sleep but when I used it on laptop my laptop, touch stopped working.

closed 2026-04-07 by github-actions[bot]