sebanc/brunch #622: Hiding Grub

massatt212, 2020-11-01:
can someone help me with a small guide on how to hide grub & the boot scrips at launch I just want it to show Chrome Screen Only.

formire, 2020-11-01:
To hide grub:
After logging in to Chrome OS
Ctrl + alt + t
shell
sudo edit-grub-config
And set grub timeout to 0

I don't know how to hide the scripts, or if you even can

sar777-dev, 2020-11-01:
Also use 'console=' kernal command line, add after 'cros_debug' with atleast one space after 'cros_debug' in the grub. To open grub - above had already stated.
then save using ctrl+o,then hit enter and exit by pressing ctrl+x.
then reboot, it will take some time only once when any change in grub file.

formire, 2020-11-01:
What do you have to type in after the console= command to disable grub?

sar777-dev, 2020-11-02:
You have to type nothing after 'console=' , left it blank or you can use framework options to activate functions like 'enable_updates'.
You should read homepage or Readme.md carefully and your required request is also stated well with example, it is located down below the webpage with headind '**Framework options**' and '**Kernal command line parameters**'.

stateandrei5, 2020-12-02:
I am also struggling with this. I have set "console=" for a no text boot but i wish for the grub menu to not be presented as well. How could i set the grub timeout to 0 as well as present a white background?

brkeejp, 2020-12-10:
take my word and use Stub loader :)
https://github.com/sebanc/brunch/discussions/699#discussion-55802

closed 2026-04-07 by github-actions[bot]