sebanc/brunch #623: Touchpad doesn't work

EddieGBU, 2020-11-02:
Hi! I have a problem with my touchpad and would like help.
I installed Chrome OS correctly on a USB to test before installing it on my hard drive, everything worked perfect there so I decided to install it on my pc, deleting Windows.
I followed the same steps as before (changing sdb to sda) and now the touchpad does not work.
I used the same files in both installations.
The click or slide does not work so it is useless.
The touchpad option doesn't even appear in settings now.
I have tried alt_touchpad_config and 2 and it doesn't fix it.
Sorry if I have bad English but I am using the translator.
I have an HP laptop and it uses the elan drivers.
Thanks in advance

sar777-dev, 2020-11-02:
I also have Hp laptop it's Hp convertible x360 and its touchpad works great than windows like in scrolling.
try to install again using new fresh files. also which method you are using to install?

EddieGBU, 2020-11-02:
I used the method that is in "code". My laptop is unfortunately not convertible :(.
What files do you think I can change? I used the latest version of brunch and version 85 of rammus.

EddieGBU, 2020-11-02:
I just looked at the ChromeOS release page and there seems to be a new version. I will try to update tomorrow. Thank you

sar777-dev, 2020-11-03:
why don't try other recovery images other than rammus like - snappy, kip, soraka, chell. 
Try to use recovery image in which hp chromebook have been made.In these hp recovery images Hp chromebook have their own 'hp' binaries which may get compatible which your hp laptop.
Right now I'm using 'snappy' recovery image as it have solved bluetooth disconnection problem which was in rammus.

EddieGBU, 2020-11-03:
Thanks for the help but I already fixed it.
I added several kernel command lines and now it works almost perfect (gestures don't work).
These are the kernel command lines that I used in case someone occupies them.
i8042.noloop
i8042.nomux
i8042.nopnp
i8042.reset
module_blacklist=i2c_hid,i8042
module_blacklist=hid_multitouch
psmouse.proto=imps
psmouse.elantech_smbus=0
psmouse.synaptics_intertouch=0
module_blacklist=elan_i2c
If any of those are bad for the pc, please tell me.

sar777-dev, 2020-11-04:
why had you added
module_blacklist=hid_multitouch
if you blacklist multitouch module, how will you use multitouch gestures( like two-finger gestures).

EddieGBU, 2020-11-04:
The truth is that I don't understand much about those commands😅. I only did what the forum told me. I'm going to remove it right now.

EddieGBU, 2020-11-04:
I removed it but the gestures still don't work. Thanks anyway.

sar777-dev, 2020-11-04:
try removing 
psmouse.elantech_smbus=0
module_blacklist=elan_i2c
see if multi-gesture works

EddieGBU, 2020-11-04:
Nope, it still doesn't work. I tried deleting module_blacklist=i2c_hid,i8042 too but it didn't work either.

EddieGBU, 2020-11-04:
I'll try another release, I'll try kip.
One question, can I use the command chromeos-update? Or it may cause some problem

sar777-dev, 2020-11-04:
I think you misunderstood something I listed the commands to be removed from the rest of your commands not to use it alone.
Actually Chrome OS cannot be installed directly – that's where branch take care of these things.
If you want to update Chrome OS you can try using framework options that is 'enable_updates' by modifying grub using cross-shell command line.
If you want help how to modify grub menu ask me.

sar777-dev, 2020-11-04:
I had also used  framework option 'enable _updates' when ChromeOS 86 was not found on the website, then it got matched with the brunch r86, also solved the  problem of high CPU usage using Android apps.

EddieGBU, 2020-11-04:
I deleted what you told me to delete. I was wondering if I could use the chromeos-update command to change releases (from rammus to kip).

sar777-dev, 2020-11-04:
that's not possible because you are trying to change rammus to kip means you are changing one device to another.
This can be done only by fresh installation.
also does touchpad gestures worked?

EddieGBU, 2020-11-04:
Gestures didn't work until I changed alt_touchpad_config from 1 to 2 but now swiping is very laggy. I prefer swiping over gestures so I will revert the changes.

sar777-dev, 2020-11-05:
Have you tried other recovery images other than rammus?

EddieGBU, 2020-11-05:
Yes, I have tried Hatch, Kip and Octopus. Only octopus made the touchpad work but the swipe is very laggy. Sometimes it detects the swipe and sometimes it doesn't.

sar777-dev, 2020-11-05:
Try framework options in 'optupus' recovery image 
"alt_touchpad_config"
"alt_touchpad_config2"
by making changes in grub menu.

sar777-dev, 2020-11-05:
Is your older or newer?
Also try chell

EddieGBU, 2020-11-05:
I think it is new, I will try the methods I used before in Octopus. If there are no results then I will try another.

sar777-dev, 2020-11-05:
Can you post 'dmesg' 
Type 'dmesg > ~/Downloads/dmesg.txt' in shell and it will save the file in Downloads folder.

EddieGBU, 2020-11-05:
[dmesg.txt](https://github.com/sebanc/brunch/files/5495733/dmesg.txt)
I discovered something. I put sudo modprobe -r psmouse and sudo modprobe psmouse elantech_smbus=1 in shell and now the touchpad works perfect, is there a way to automate that command?

sebanc, 2020-11-05:
Yes, adding "psmouse.elantech_smbus=1" on the kernel command line (after cros_debug) should do the same.

EddieGBU, 2020-11-05:
Wait a sec, i'll try

EddieGBU, 2020-11-05:
Thank you! Everything works perfect, gestures, clicks, swipe, everything!

sar777-dev, 2020-11-05:
That's good news to hear.

EddieGBU, 2020-11-16:
Hello, it's me, again. The touchpad just crashed again, I tried removing the kernel command and putting it back in and it doesn't work.

EddieGBU, 2020-11-16:
I'd better go back to Windows. Very good project but unfortunately my laptop is very incompatible. I liked the system so I'll be back when I get another device. Bye

closed 2020-11-16 by EddieGBU

sar777-dev, 2020-11-16:
See you soon

bybenjamin, 2021-10-06:
> Yes, adding "psmouse.elantech_smbus=1" on the kernel command line (after cros_debug) should do the same.

The system works perfectly. Thank you for the project. Touchpad stops working after suspend. What I did to get it running again was '' sudo rmmod i2c_hid ; sudo modprobe i2c_hid '' How do I prevent the i2c_hid module from suspending Thanks in advance for your help

AkhmadFaiz012, 2024-02-26:
> Yes, adding "psmouse.elantech_smbus=1" on the kernel command line (after cros_debug) should do the same.

Thank you so much this work perfectly, anyway can u explain why when i plug charge in my laptop the battery icon not change and still same but the percentage still up, thanks