sebanc/brunch #625: stuck at writing partition 3

nonregla1, 2020-11-02:
installing on a lenovo ideapad 120s-11iap. 
keeps sticking at writing partition 3. System completely frozen.
Tried using rammus, samus and grunt. tried >1 usb stick.
any ideas? 
I would love to get this to work!!

sar777-dev, 2020-11-02:
why don't you try live linux style as described in video of 'ETA Prime' on youtube, its very easy and I had been successful every time.
link:- https://www.youtube.com/watch?v=ROBpXNonVxc

sebanc, 2020-11-04:
@sar777-dev Even though it worked for you I would prefer that youtube videos are not referenced as they often rely on unsupported scripts.

@nonregla1 Try again after rebooting your computer.

sar777-dev, 2020-11-04:
@sebanc could you please explain what's wrong in his video.

sebanc, 2020-11-04:
The use of this script:
https://raw.githubusercontent.com/shrikant2002/ChromeOS/master/install.sh

which hardcodes /dev/sda as destination whereas it could be an hdd, an usb drive or nothing.

Then people come here and say it does not work.

sar777-dev, 2020-11-04:
Before installing look at the disk name by using 'Disks', since we are live booting linux by linux bootable pen drive.
I had install brunch ChromeOS on my nvme drive by changing the script file /dev/nvme0n1 on external ssd by changing script /dev/sdb.
Thats all, just simple changes but you are right people don't pay attention then come and say it didn't work.