sebanc/brunch #629: Can I override DSDT using cpio archive or other method?

nekomimipanic, 2020-11-03:
My tablet's ACPI table is different from standard one.
It need to fix ACPI table, otherwise not work volume and power button.
Ref. https://github.com/ivzave/matebook-linux

 Is there any  easy way to override DSDT before boot?
(already prepared recompiled DSDT as dsdt.Aml)
Cpio method can't work.
Ref. https://wiki.archlinux.org/index.php/DSDT
Ref. https://android.googlesource.com/kernel/tegra/+/ce636452343af0522d2666157dab9c2096f4f996/Documentation/acpi/initramfs-add-dsdt.sh

I recompile kernel,but it's annoying and takes time.
So is there any other way to override DSDT?

Thx

sebanc, 2020-11-04:
If it is just about the hardware buttons, it can be patched directly in the kernel, instead of using DSDT.

Could you attach a dmesg to this issue ?

nekomimipanic, 2020-11-06:
Here is dmsg log.
It's DSDT integrated kernel one.
[dmesg2.log](https://github.com/sebanc/brunch/files/5503875/dmesg2.log)

sebanc, 2020-11-07:
I have added you to the brunch testing repo (you should have received a github invite), the release there should have working buttons for your device.

nekomimipanic, 2020-11-08:
power button is working testing brunch.
adittionaly this device bluetooth firmware is not properly adopt.
need root-c mount and cp BCM4356A2-0a5c-6419.hcd BCM4354A2-0a5c-6419.hcd

finally rotate is not work this testing one
force_tablet_mode or sysfs_tablet_mode or already mentioned method not working.

thx

sebanc, 2020-11-08:
I have a doubt as you mention tablet mode:
- Is tablet mode correctly triggered but the device does not rotate ?
- Or is tablet mode not triggered ?

nekomimipanic, 2020-11-08:
May be tablet mode not triggered.
on sysfs_tablet_mode,
# echo 1 > /dev/pra~
but no action.

before testing update, it worked.

sebanc, 2020-11-08:
Sorry but I really need to confirm that it is a tablet mode issue or sensors.

Could you remove all tablet mode related options, reboot, login then press CTRL+ALT+SHIFT+T to trigger tablet mode manually and see if rotation works ?

nekomimipanic, 2020-11-08:
CTRL+ALT+SHIFT+T , no reaction...

sebanc, 2020-11-08:
I have uploaded another brunch-testing, could you try it ?

Between, do you remember which release your were using previously (when tablet mode worked fine) ?

nekomimipanic, 2020-11-09:
20201109 not working rotation
and ctrl~ notworking too.
force~ not working.
r86 worked fine.

thx

sebanc, 2020-11-09:
Well, the issue is I did not change much between r86 and this build so I am not sure what it could be related to...

Could you attach a dmesg to this issue ? (after a fresh reboot, run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your Downloads folder to this issue)

sebanc, 2020-11-09:
Forgot to say, a new dmesg without any tablet mode related options

sebanc, 2020-11-10:
You can also try the new testing release

nekomimipanic, 2020-11-10:
new one 201110ver work fine!
thx.

ill attach dmesg log with 09 and 10.
[dmesg.zip](https://github.com/sebanc/brunch/files/5518052/dmesg.zip)

nekomimipanic, 2020-11-10:
Hmm.
both testing from  chromeos-install.sh,not work. (clean install)
Stable can rotate,

so first stable chromeos-install.sh then,chromeos-update -f 201110ver  can rotate...
stable chromeos-install.sh then,chromeos-update -f 201109ver  NOT rotate.

[stable_20201014dmesg.log](https://github.com/sebanc/brunch/files/5518327/stable_20201014dmesg.log)

sebanc, 2020-11-10:
Thanks for all the tests :) that's very interesting (even though I don't really understand yet what the issue is)

nekomimipanic, 2020-11-11:
can I check your additional <<ACPI>> code?
I ll test step by step.

sebanc, 2020-11-11:
You will find my patch below but I doubt it is related to the tablet mode issue.

```
--- a/drivers/platform/x86/intel-hid.c	2020-04-20 19:28:32.000000000 +0200
+++ b/drivers/platform/x86/intel-hid.c	2020-05-04 18:01:26.427423389 +0200
@@ -60,6 +60,20 @@
 
 static const struct dmi_system_id button_array_table[] = {
 	{
+		.ident = "HUAWEI MateBook HZ-W09",
+		.matches = {
+			DMI_MATCH(DMI_SYS_VENDOR, "HUAWEI"),
+			DMI_MATCH(DMI_PRODUCT_NAME, "MateBook HZ-W09"),
+		},
+	},
+	{
+		.ident = "HUAWEI MateBook HZ-W19",
+		.matches = {
+			DMI_MATCH(DMI_SYS_VENDOR, "HUAWEI"),
+			DMI_MATCH(DMI_PRODUCT_NAME, "MateBook HZ-W19"),
+		},
+	},
+	{
 		.ident = "Wacom MobileStudio Pro 13",
 		.matches = {
 			DMI_MATCH(DMI_SYS_VENDOR, "Wacom Co.,Ltd"),
```

sebanc, 2020-11-12:
Could you try the new "brunch-testing" release ?

nekomimipanic, 2020-11-12:
testing release not work rotate.
thx

sebanc, 2020-11-12:
Could you attach an "lsmod" to this issue ?

nekomimipanic, 2020-11-12:
here is files
you have a chanse to build stable-r86  with only mod hz-w19 in intel_hid.c ,I want to try it.

[dmesg.log](https://github.com/sebanc/brunch/files/5532148/dmesg.log)
[lsmod.log](https://github.com/sebanc/brunch/files/5532150/lsmod.log)

sebanc, 2020-11-13:
The MateBook HZ-W09 part exist in brunch for more than 6 months so that's not related.

Actually, I just noticed that there are 2 tablet mode modules loaded on your device (which chromeos does not expect), could you try to add "module_blacklist=intel_vbtn" on the kernel command line (after "cros_debug")

nekomimipanic, 2020-11-13:
no change
[lsmod.log](https://github.com/sebanc/brunch/files/5537316/lsmod.log)