sebanc/brunch #630: HP Spectre X360, sleep no longer working and only IR camera

tduong-cloud, 2020-11-05:
Hi Sebanc

Many thanks for your great work. I have been able to install and enjoy ChromeOS on my HP Spectre 13" X360 I have bought in 2018.

I used eve recovery and so far the only issues I have really encountered are :

- Camera: I can only have access to the IR camera, not really useful

- Sleep: I was able to make it work using the options in grub : suspend_s3, acpi_power_button. I have also disabled TPM. It was working fine until I have updated ChromeOS 85 to 86. I have updated brunch. And now sleep is no longer working as the screen freezes and stay black.

Do you have any ideas on those two issues?

Looking forward to hearing from you.

Best

sebanc, 2020-11-05:
Hi,

For camera, using samus recovery image should probably help.

WRT suspend, I have not change suspend related features in a long while, have you received a bios update in windows recently ? In any case I am sorry to say those issues are really hard to solve.

Also if you have enabled the option "mount_internal_drives", it is known to break suspend.

tduong-cloud, 2020-11-08:
Hi,

Already have tried samus, rammus, and even other recovery and it did not work for the camera.

Concerning the suspend features, I don't know about the bios, but I don't have the "mount_internal_drives" option enabled. I guess I would need to find another solution on this. The last one, disabling TPM worked fine, but now I don't know why it doesn't work anymore.

Again, many thanks for your great work !

Best

sebanc, 2020-11-11:
Actually your laptop is recent enough so I added you to the "brunch-next" private repository (which has a more recent kernel). You should have received a github invite.

Not sure if the more recent kernel will help with your issues but it's probably worth trying it.

tduong-cloud, 2020-11-11:
Thanks, I will try it.

tduong-cloud, 2020-11-13:
I have just updated brunch with the brunch-next version and it looks like it is working again ! Many many thanks for your great help ! I love chromeos!