sebanc/brunch #631: How to modify kernel parameters in native Chrome init?

daviddean99, 2020-11-05:
I've installedchromeos_13421.73.0_samus_recovery_stable-channel_mp-v3.bin with brunch_r86_k4.19_stable_20201014 onto my 2014 Toshiba Chromebook 2 (swanky).

I did this by booting up using Linux Mint on a USB, then using the chromeos-install.sh script targeted at the chromebooks local mmc storage.

This seems to have worked - it's booting up in Developer mode no problem, and seems to be using the native Chrome init because I don't see a grub splash page.

The problem is that I need to add some optional kernel parameters for the bay trail processor features and to improve trackpad performance.

But how/where do I add those options if the native Chrome init is being used instead of grub?

There doesn't seem to be a grub-edit command either.

Thanks for your help.

keweminer, 2020-11-05:
For editing grub? Have you tried "sudo edit-grub-config" is that what you're looking for?

sar777-dev, 2020-11-05:
Is installed ChromeOS is running or your old ChromeOS is running?

daviddean99, 2020-11-05:
It appears what had happened is that the stock ChromeOS bootloader was starting up, then executing the recovery image that I'd written to the hard drive using the chromeos-install.sh command, i.e. latest ChromeOS, stock bootloader.

None of the brunch parameters were being executed because the grub bootloader wasn't launched.

So, I've now completely replaced the stock ChromeOS bootloader and it's running grub as expected. I can put in my kernel parameters on startup and the mouse and audio is working perfectly.

It's not as good as using the stock ChromeOS bootloader due to the longer startup time, but it means I can use the latest ChromeOS on this old (but still capable) 2014 Chromebook. Awesome!!

closed 2020-11-05 by daviddean99

sar777-dev, 2020-11-05:
That's good news to hear.