sebanc/brunch #632: CHROME OS starts well, but I can’t login into my gmail account

Hujun574, 2020-11-05:
Hi All,
I’ve a problem with chromebook login, and I can’t fix it by myself;
Machine info: CPU G2030 INTEL, 8G DDR3 Memories, harddrive 500g Hitachi, no wireless hardware.
I installed chrome os(recovery image is samus 85) as per the video that Kedar Nimbalkar, a YouTube contributor, offered.
It runs well, except I can’t login into my gmail account.
The system keeps on telling me the internet is not linked, but I do have the Ethernet settings done properly.
I wonder whether I must buy a wireless hardware, but I don’t think so logically.

sar777-dev, 2020-11-05:
Are taking about email website Gmail?

Hujun574, 2020-11-05:
> Are taking about email website Gmail?

No, it's a starting page of chrome os.
I mean to login page.
Os asks you to login your google account to continue......

sar777-dev, 2020-11-05:
You are right, it didn't have to do anything with connectivity, your Ethernet connection is enough.
Can it have the option to run in guest mode?

sar777-dev, 2020-11-05:
why don't you try brunch r86 and chromeos recovery images version 86 as 86 recovery image is now available.

sebanc, 2020-11-05:
Could you try to press "e" when you see grub and add "module_blacklist=tpm_tis" to the kernel command line (after "cros_debug")?

If it does not help, try to clear chrome synced data (fom your phone or a linux live usb):
- Login to your Google dashboard account
- Go to the "Chrome" application -> Manage synced data
- Click on "reinitialize synced data"

Hujun574, 2020-11-06:
> why don't you try brunch r86 and chromeos recovery images version 86 as 86 recovery image is now available.

Really, Version 86?  Tks, Mate!
I will be glad to have it tried shortly.

Hujun574, 2020-11-06:
> Could you try to press "e" when you see grub and add "module_blacklist=tpm_tis" to the kernel command line (after "cros_debug")?
>
> If it does not help, try to clear chrome synced data (fom your phone [...]

noted. Tks.
I will try in these 2 ways also.

Hujun574, 2020-11-06:
Onething I gotta to tell you all that's pretty weird. When I pretend to forget my password, so google system asks you to imput the auxiliary mailbox add, then I follow. Incredibly the code is sent to this mailbox indeed, though I could not login either.

Hujun574, 2020-11-06:
> You are right, it didn't have to do anything with connectivity, your Ethernet connection is enough. In fact my other Dell pc is also using connectivity through Ethernet.
> Can it have the option to run in guest mode? I guess not though.

Yes, you're right. Can't run in guest mode either.

sebanc, 2020-11-06:
I think I misunderstood this issue, your issue is that your ethernet adapters does not appear as connected ?

If so, your ethernet card driver is probably not enabled in the kernel config and I would need either the ethernet card model or the output of "lsmod" in a linux distro.

Hujun574, 2020-11-07:
As to the ethernet card model, surely I can provide when I go back to the office next Monday. But what does the output of "lsmod" in a linux distro mean? Should I type SUDO ISMOD in linux mode(black background with white wordings)? Honestly, I'm a pure rookie regarding any linux platforms, not include the grapic integrated linux os like linux mint or pop os. I mean to something like the starting step as OPENWRT's, the codes runs everything time you start or restart it.......

sar777-dev, 2020-11-08:
you have to type lsmod( like l in letter not i ) in shell by typing 'shell' command first in command line and then 'lsmod'.
It displays kernal which kernal modules are loaded.

Nyta8591, 2023-06-02:
I have the same issue with my HP Chromebook 14 4G [(,](https://chromebookdb.com/chromebook/hp-chromebook-14-14-ak040nr/) tried Kernel 15.15 and 15.10, but it keeps getting stuck at Google Login page and no wifi in Guest mode (doesn't load anything on Chrome even though says Wifi connected sucessfully). I also tried loading the latest r111 recovery [https://github.com/sebanc/brunch/releases/tag/r111-stable-20230414](url) through sudo chromeos-update -r/recovery.bin by mounting USB drive, it was installed sucessfully, but didn't help..... 

Also the audio and bluetooth won't work on this device, even though though I tried activating "chromebook_audio" via Framework options. I managed to extract Audio drivers from original ChromeOS, can I mount them somehow into Brunch???

Here are my lsmod outputs from original ChromeOS. Can someone help plssss? Thanks so much!!!
[lspci, lsusb, lsmod.txt](https://github.com/sebanc/brunch/files/11636496/lspci.lsusb.lsmod.txt)