sebanc/brunch #633: invalid opcode in libhoudini.so (86.0.4240.183)

daviddean99, 2020-11-05:
I'm trying to get Roblox (and other apps!) working for my daughter on an old Chromebook that's been successfully updated to a newer ChromeOS using Brunch.

The problem is that when I launch Roblox it throws this error:

`INFO kernel: [  418.436298] traps: Thread-6[6387] trap invalid opcode ip:792c52e740cb sp:792c3cada008 error:0 in libhoudini.so[792c52d98000+4f2000]`

The processor is an Intel Celeron N2840 from the BayTrail chipset.

I've tried both rammus and samus (13421.73.0) images, and currently using brunch_r86_k4.19_stable_20201014.

I'm guessing the problem is happening with Android apps which only have ARM support and don't include x86, hence the libhoudini.so error?

Would really appreciate any suggestions on how to fix this problem? I can't find anything helpful on Google.

Thanks!

_Full processor info:_

```
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 55
model name	: Intel(R) Celeron(R) CPU  N2840  @ 2.16GHz
stepping	: 8
microcode	: 0x829
cpu MHz		: 1636.295
cache size	: 1024 KB
physical id	: 0
siblings	: 2
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer rdrand lahf_lm 3dnowprefetch epb pti tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat
bugs		: cpu_meltdown spectre_v1 spectre_v2 mds msbds_only
bogomips	: 4331.60
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:
```

sebanc, 2020-11-05:
Hi,

If I am not mistaken libhoudini is responsible for arm apps compatibility and it visibly requires cpu instructions that your cpu does not have ("invalid opcode" message).

I was not aware of that limitation and as it is related to the android container there is not much I can do.

You can try "reef" or "coral" recovery images which are closer to your device cpu (still both have more recent cpu so I am not sure it will help)

daviddean99, 2020-11-07:
It seems that Chromebooks using Intel’s BayTrail do not include VT-x (Intel Virtualization Technology):

https://chromium.googlesource.com/chromiumos/docs/+/master/containers_and_vms.md#baytrail-systems

Could it be possible that libhoudini.so relies on VT-x support to run ARM binaries on x86, and so with my CPU not supporting VT-x it throws the opcode error that I'm seeing?

daviddean99, 2020-11-07:
Answering my own question for the benefit of others - yes, VT-x is required for libhoudini.so to run ARM binaries on x86.

You can install Google Play, but if an application only includes ARM libraries and not x86 (like Roblox) then it simply won't run on the following hardware boards:

banjo
candy
clapper
enguarde
expresso
glimmer
gnawty
heli
kip
ninja
orco
quawks
squawks
sumo
swanky
winky

Oh well, looks like I'll have to get a more up-to-date device :(

closed 2020-11-07 by daviddean99

kin-kwok, 2020-11-14:
I got the same problem since about a month ago on my Acer C740 Paine which has a Inter 3205U Broadwell CPU.  Same libhoudini.so opcode traps when starting Roblox.  Maybe Roblox did change some of their build to stop some old broad working.