sebanc/brunch #634: lots of lag while using

zainullahkj, 2020-11-08:
lots of lag while using chromeos its a gpu problem is there a fix

sebanc, 2020-11-12:
You can try the latest version in the brunch-testing repo (an invite has just been sent) and also add "enforce_hyperthreading=1" on the kernel command line (after cros_debug)

lupohirp, 2020-11-13:
Sebanc i've the same problem after a lot of use. Can you invite also me?

sebanc, 2020-11-13:
@lupohirp Sure, invite sent

lupohirp, 2020-11-14:
> @lupohirp Sure, invite sent

Thanks sebanc. Where I can open issues on testing? Bluetooth it's not working anymore with latest 5.4 kernel, instead with the latest one of brunch next was working.

sebanc, 2020-11-14:
Could you post a dmesg in this issue ?

lupohirp, 2020-11-14:
Sure, here it is.

[dmesg.log](https://github.com/sebanc/brunch/files/5540641/dmesg.log)

sebanc, 2020-11-14:
I think it's an issue due to the kernel version I use for those releases (r85) against r87 on brunch-next. I will update the kernel version soon, meanwhile you can try "iwlwifi_backport" option.

lupohirp, 2020-11-14:
with that option, my wifi doesn't work anymore....

sebanc, 2020-11-22:
@lupohirp I uploaded a new testing build which should fix your bluetooth issue.

lupohirp, 2020-11-23:
> @lupohirp I uploaded a new testing build which should fix your bluetooth issue.

Confirmed! Bluetooh now works! thanks!

jhs1004, 2021-01-27:
Hi, @sebanc, I am currently using a Macbook Pro 15 2013 late with iGPU Iris Pro and I believe I have a GPU compatibility problem as I get loads of lags and stutter. I am currently on R87 Brunch stable and corresponding Rammus image.Do you think trying out R81 with maybe Samus may help with the GPU problem I believe I have?

khanhtranngoccva, 2021-02-28:
If you use an HDD, grab an SSD, attach it however you can, and be done :)