sebanc/brunch #636: No menuentry appears when installing through  ubuntu wsl

ameyjoshi3068, 2020-11-08:
no error message was shown.
[image]

Is this happening due to limitation of wsl?(sandboxed architecture)

sebanc, 2020-11-08:
I disabled config generation for WSL1 because it has no notion of /dev/xxxxx for "img_part" parameter.

For WSL2, I have no idea if the config displays, from my understanding it should have /dev/xxxxx but I am not sure if the config displays.

ameyjoshi3068, 2020-11-08:
So, for this situation I think there should be a template boot configuration in readme
I will submit a pull request soon. You can merge it if you want.

sebanc, 2020-11-08:
Wait a sec, if you remove:
`grep -qi 'Microsoft' /proc/version || `

from chromeos-install.sh it will display the config, could you check if the generated config with WSL2 is correct ?

If it is the case, we can just remove that and add WSL2 as a requirement.

ameyjoshi3068, 2020-11-08:
Yes, menuentry is generated but there is drive letter instend of device path after img_part

menuentry "ChromeOS (boot from disk image)" {
        rmmod tpm
        img_part=D:\
        img_path=/brunch_r85_k5.4_20200922/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}

sebanc, 2020-11-08:
Indeed that's what happened with WSL1 also, I don't think we can get the correct "img_part" from WSL.

Maybe we should change the guide so that:
- The user creates a chromeos usb flashdrive
- creates the dual boot image in chromeos
- Saves the config to a file that can be used to create the grub2win config from windows

Someone more experienced can still understand how to take a shortcut.

ameyjoshi3068, 2020-11-09:
Can we think 🤔 of using linux from live usb  for the image creation?. 
It would also give access to other linux tools which can also be used to identify exact target device.

sebanc, 2020-11-09:
In my opinion, even though I also think ext4 is better, the basic instruction set for Windows should not request the user to create partitions. We can still add an entry in the wiki for those who might want to create dedicated ext4 partitions.

Also, most users will have already a brunch usb flashdrive made to try it from usb according to the previous step so I think it would be better to just re-use this usb flashdrive.

My suggestion would be that:
- When creating a dual boot image with "chromeos-install", additionally to the grub config being displayed on the screen, I save the grub config text in the same folder as the dual boot image with `<dual_boot_image_name>.txt` as name.
- The user then just has to go back into windows and copy the text from `<dual_boot_image_name>.txt` file

What do you think ?

ameyjoshi3068, 2020-11-09:
I really liked your idea regarding the text file.
It will surely be helpful.

sebanc, 2020-11-09:
I have sent you an invite to brunch-testing branch. If you want to test, the release there generates a `<dual_boot_image_name>.grub.txt` file

ameyjoshi3068, 2020-11-09:
Thanks sebanc, I will surely test it

closed 2020-11-09 by ameyjoshi3068