sebanc/brunch #638: Acer C720 Bluetooth Fail

Nunthewiser, 2020-11-09:
Brunch v86, ChromeOS (Rammus) V86

I know it's really not a new issue and it's probably a ChromeOS issue anyway.

 Bluetooth connects (sometimes) but continues to play through internal speakers.  I have tried 4 different sets of earbuds and speakers. I have tried the commands in crosh, still nothing.  Works just fine in Linux.

Otherwise, it is great (thanks Sebanc!).  Gives the old C720 a new lease on life.  I have a similar EOL Dell  first generation chromebook that I would love to update, but I hate to lose bluetooth.

Any suggestions?

sebanc, 2020-11-09:
Could you post attach a dmesg to this issue ? (after a fresh reboot, run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your Downloads folder to this issue)

Nunthewiser, 2020-11-09:
Something is a little bit wrong with that.  I get "error: Unknown option" when I paste that.  Plain "dmesg" spits out a whole lot of stuff that I guess you are trying to get me to capture, but it doesn't like that exact command.

sebanc, 2020-11-09:
Press CTRL+ALT+T to open crosh, then type "shell" and then the command I provided

Nunthewiser, 2020-11-09:
[dmesg.txt](https://github.com/sebanc/brunch/files/5512128/dmesg.txt)

Nunthewiser, 2020-11-09:
Sorry.  I am dense.  Did that work?

sebanc, 2020-11-09:
yes it is perfect, however I don't identify a specific issue with it.

I am going to try something in the next release

Nunthewiser, 2020-11-09:
OK.  Thanks.for brunch and good luck.

sebanc, 2020-11-10:
You should have received a github invite to "brunch-testing" repository. Could you validate the invite and try the release there ?

Nunthewiser, 2020-11-10:
Monsieur Sebanc-

I am truly honored to be chosen to participate in testing.  I am very
unskilled, though.  I usually get lucky and can follow guides and
directions after some mistakes.

I must confess that  I got brunch working on my C720 by installing using
Kadar Nimbalkar's video.  Can I just try to reinstall ChromeOS v86 using
the new test brunch file? (That I can do) I really don't know anything
about linux commands , etc, though I do have access to Linux Mint and
Windows on another computer.

Rick

sebanc, 2020-11-10:
yes if you could try to reinstall using the same procedure but with the testing brunch release and see if there is any improvement that would be perfect

Nunthewiser, 2020-11-10:
I can do that.  To be honest, what happened to me and what seems to
happen to many people who try this using the "Kedar" method is that the
installation script, when downloaded originally using windows and extracted
using winzip, changes the file and inserts "/r" after every line.  I tried
to remove them using notepad ++ but it didn't work for me.  So , I just
downloaded and extracted all in linux and then it worked well.  I will just
do that again.

Nunthewiser, 2020-11-10:
OK , I installed the test file, but, unfortunately,  I don't see any improvement.  ChromeOS struggles to connect to the two devices.  I turn on/off/pair/unpair, etc.  Eventually ChromeOS says it is connected, but the devices still look like they are trying to pair.  Audio still comes from the internal speakers.  Switching to Linux Mint, both devices pair instantly and work.  I downloaded the new dmesg, attached.

[dmesg.txt](https://github.com/sebanc/brunch/files/5520477/dmesg.txt)

sebanc, 2020-11-11:
Could you download the below file, then open crosh shell (CTRL+ALT+T and type "shell" at the invite) and then run:
`sudo tar zxvf ~/Downloads/ath3k.tar.gz -C /lib/firmware`

[ath3k.tar.gz](https://github.com/sebanc/brunch/files/5522727/ath3k.tar.gz)

sebanc, 2020-11-11:
Forgot the last part: Then reboot and test bluetooth

Nunthewiser, 2020-11-11:
OK, did that
[image]

This did not seem to make any difference.  The two devices and the C720 behave the same.  This same issue happens when I try to connect to a bluetooth keyboard- the C720 says it's connected but the device looks like still in pairing mode.

mikka9, 2020-11-11:
I have the same issue on my c720
[dmesggggg.txt](https://github.com/sebanc/brunch/files/5524726/dmesggggg.txt)

sebanc, 2020-11-11:
Could you try the new brunch-testing build ? It has a more recent kernel

mikka9, 2020-11-11:
Sorry, I have the latest build, isn't it?

Nunthewiser, 2020-11-11:
I think he was talking to me, Nunthewiser.  I am working on something that
was built about 30 minutes ago...should have an update in a few minutes to
see if it improves.

Nunthewiser, 2020-11-11:
That works!  Both bluetooth earbuds connected quickly and played normally.. Typing this on my Anker bluetooth keyboard, also working.  Thanks.

sebanc, 2020-11-11:
Nice :)

@mikka9 I did not notice pay attention that there were several users in this issue, you should have received an invite to brunch-testing. Try the release there.

Nunthewiser, 2020-11-11:
Sebanc-  As I said upthread, I have a 1st generation Dell Chromebook 11 CB1C13 with the same 2955u Celeron as the C720 that I was going to update.  Can I try the latest test on that one?  Or I can do some experimenting on it if you like.

mikka9, 2020-11-12:
> Nice :)
> 
> @mikka9 I did not pay attention that there were several users in this issue, you should have received an invite to brunch-testing. Try the release there.

Great. 5.4 works fine!

sebanc, 2020-11-12:
@Nunthewiser Sure you can try it on your Dell, I am probably going to switch to 5.4 as main kernel soon so I need as much feedback as possible

Nunthewiser, 2020-11-12:
I installed the 111120 test build on Dell Chromebook 11 CB1C13 (WOLF).  Everything, including bluetooth seems to work well.

sebanc, 2020-11-12:
That's good news :) Thanks for the feedback !

mikka9, 2020-11-12:
I was too optimistic
c720 connected to bt speaker one or two times, but sounds interupted time to time. But now it can't connect again(((
I removed paired device and paired again. Listened to the music 2min and speaker disconnected again......
[dm2.txt](https://github.com/sebanc/brunch/files/5532847/dm2.txt)

Nunthewiser, 2020-11-12:
My C720 bt seems ok today.  I tested 2 sets of earbuds.  The first worked well.  The second didn't connect well and would only work for a few seconds before disconnecting.  So, I rebooted and the second pair seems to work well.

Nunthewiser, 2020-11-17:
Update on the OP C720. I have been using it for about a week on 4 different bluetooth earbuds and speakers.  It works well, basically.  Sometimes it seems to get confused when switching devices, but a reboot solves it.  It never drops out or stutters when connected.  I can usually go 10 meters away before it fades.  (The dell is working well on BT as well).  I hope you use this kernel in the next drop (?).

AryanRai, 2020-11-29:
> Press CTRL+ALT+T to open crosh, then type "shell" and then the command I provided

What is the command you provided. I am also having the same bluetooth issue on r86