sebanc/brunch #640: "Sorry, your password couldn't be verified" in dual boot.

dhyogobr, 2020-11-09:
Hi, I have this problem after first setup in dual boot, l can't log in after reboot.
In USB Flash drive chromeos works fine.

[install_log.txt](https://github.com/sebanc/brunch/files/5512912/install_log.txt)
[dmesg_ssd_1st_boot.txt](https://github.com/sebanc/brunch/files/5512914/dmesg_ssd_1st_boot.txt)
[dmesg_guest_2nd_boot.txt](https://github.com/sebanc/brunch/files/5512915/dmesg_guest_2nd_boot.txt)
[dmesg_usb_working.txt](https://github.com/sebanc/brunch/files/5512916/dmesg_usb_working.txt)

sebanc, 2020-11-10:
Hi, you should have received a github invite to "brunch-testing" repo. Could you validate the email invite and try to install the release there on your hdd ?

dhyogobr, 2020-11-10:
Hi, I downloaded brunch-testing, formated partition, reinstalled chromeos, but problem persists, when I boot first time in ssd, always have this erros in 30%, when I boot first same img from pendrive this bar have 100% without stop with 30%. 
[image]

sebanc, 2020-11-10:
Unfortunately, it is just some text displaying while the copy operation is going but it is not an error.

On which type of partition did you create the dual boot image ? (ext4 or ntfs)

dhyogobr, 2020-11-10:
I tried with ntfs, ext4 and ext2.

sebanc, 2020-11-10:
Actually I get it now, the line you point at is responsible for the issue but not when rebuilding the system, on subsequent boots when it starts too late...

So globally it seems to mean that your system does not generate entropy fast enough to initialise /dev/{u}random early.

In theory this should be fixed by adding "random.trust_cpu=on" on the kernel command line, could you try it ? (we have already tried with "random.trust_cpu=1" which should have the same effect but I don't understand why it did not work)

dhyogobr, 2020-11-11:
I tried "random.trust_cpu=1" and "random.trust_cpu=on" but without sucess. I believe usb flash loads only 20MB/s and have time to "generate entropy" while ssd loads 450MB/s. What do you think?

sebanc, 2020-11-11:
Could you try the new release I posted in brunch-testing ? I added features so that entropy generates faster

dhyogobr, 2020-11-12:
Tested Brunch r86 k5.4 unstable 20201112 but same problem. :(

sebanc, 2020-11-12:
Could you post a new dmesg ?

dhyogobr, 2020-11-12:
Yes, here. (generated in guest login)
[dmesg.txt](https://github.com/sebanc/brunch/files/5532542/dmesg.txt)

sebanc, 2020-11-12:
I have significantly improved entropy generation in the kernel (verified on my own device) but it appears not to have any effect on your device for some reason I don't really understand. Before I look into more complex methods to generate entropy, could you check in your bios if there is some setting regarding "CRNG" "Randomization" "Random number generation" or something like that ?

dhyogobr, 2020-11-12:
Here my all bios options: (thanks for support)
[B_Opt.zip](https://github.com/sebanc/brunch/files/5532670/B_Opt.zip)

DrFunkenbreakz, 2020-11-12:
Just want to say Im having the same/similar issue here on a 4th gen i7 laptop booted from SSD (multiboot). Sometimes I can login, sometimes I type my password and it just hangs on the login screen..no error is displayed...and after a while the password ive typed clears but it stays on the login screen. No password can then be re-entered or guest mode used. Only way out is to shutdown by the shutdown button and then boot and try again. Looks like some timing issue as you have discussed above.

Due to covid and limited time, i've not had chance to look into logs yet. I will keep an eye on this thread though.

By the way, thank you so much for this excellent project. It's amazing!!

sebanc, 2020-11-12:
@DrFunkenbreakz I have just sent you a github invite to brunch-testing, even though the situation did not improve for @dhyogobr, I would like to know if there is some improvement on your side.

DrFunkenbreakz, 2020-11-12:
Thank you. I will test when back home this evening and report back on the issue thread.

Thanks once again for letting us use your hard work and project. Very much appreciated.

DrFunkenbreakz, 2020-11-12:
Updated brunch to latest test (Brunch r86 k5.4 unstable 20201112) and it appears to have fixed my login issues. Just tried 5 times and worked each time with no hang on the login screen. Will continue to test and report back tomorrow. Thank you

sebanc, 2020-11-12:
@DrFunkenbreakz thanks for the feedback :) hope it stays like that

@dhyogobr Could you try to boot with "brunch_timeout=30" added on the kernel command line (after "cros_debug") ? (it is not a definitive solution but it might be a temporary workaround if it works)

dhyogobr, 2020-11-13:
"brunch_timeout=30" not work too :(

sebanc, 2020-11-13:
@dhyogobr I added "haveged" in the new testing release however I am not sure if it is properly setup... Could you try it when you have the time and post a dmesg ?

dhyogobr, 2020-11-13:
Hi @sebanc, downloaded 20201113 version, installed, password acepted after reboot but very slow, after enter password login works after 3~5 minutes. 
[dmest_1stboot.txt](https://github.com/sebanc/brunch/files/5539674/dmest_1stboot.txt)
[dmesg_2ndboot.txt](https://github.com/sebanc/brunch/files/5539675/dmesg_2ndboot.txt)

sebanc, 2020-11-14:
Actually haveged did not work, could you try the new testing release ?

dhyogobr, 2020-11-14:
Hi @sebanc , have greatest news, after instal 20201114 version, rebooted 3 times, and login working fine! :)
Thank you very much! For your time, effort and support! Another stable versions will they have this fix?

Model: HP1000-1230BR
Processor: Intel Core i3-2328M 
Recovery: chromeos_13421.73.0_kip_recovery_stable-channel_mp-v3.bin

[image]
[dmesg_1stboot_20201114.txt](https://github.com/sebanc/brunch/files/5540720/dmesg_1stboot_20201114.txt)
[dmesg_2ndboot_20201114.txt](https://github.com/sebanc/brunch/files/5540721/dmesg_2ndboot_20201114.txt)

sebanc, 2020-11-14:
ah indeed that's great news :)

It seems haveged worked this time. Surely this fix will fix will be included in the future stable releases.

closed 2020-11-14 by dhyogobr