sebanc/brunch #644: Wifi driver

koskr29, 2020-11-14:
Hi Sebanc! 
First I want to thank you for this amazing work!

Second, I want to know how to eneable the wifi card or how to add it's driver to the system. The wifi card is the realtek rtl8723de, I found a driver (https://github.com/smlinux/rtl8723de) for linux but I don't know how to make it works.
Thank you again (:

sebanc, 2020-11-14:
Hi,

This driver is already included.

If you followed the single boot procedure:
- run "sudo edit-grub-config" in crosh shell (CTRL+ALT+T and type "shell") and add a space and "options=rtl8723de" after "cros_debug"
If you followed the dual boot procedure (disk image):
- Modify the grub config you have created in windows or linux and add a space and "options=rtl8723de" after "cros_debug"

koskr29, 2020-11-17:
It worked! Thank you 👍