sebanc/brunch #645: Couldn't connect to blutooth devices

Zarvis12, 2020-11-14:
I've installed Rammus v86 on my HP Notebook - 15-ac122tu works fine 
Wifi works but Blutooth keeps on searching but couldn't show any devices nor I can't find this laptop on my phones.

[image]

raytrix71, 2020-11-14:
did you enabled options ? such as your wireless card ?

sebanc, 2020-11-14:
I am currently extending the tests on 5.4 kernel which seems to improve bluetooth issues. I have just sent you an invite to "brunch-testing" repo which has a 5.4 kernel build, could you try it ?

A little reminder just in case, phones cannot be paired in chromeos as there is a dedicated menu.

Zarvis12, 2020-11-15:
I'm using some apps that are important and preparing for exams right now in the laptop. I'm fine for now without bluetooth and I just wanted you to know that this isn't working. If I could I will do it after 2021 May.

Zarvis12, 2020-11-15:
> did you enabled options ? such as your wireless card ?

from where?

Zarvis12, 2020-11-15:
> did you enabled options ? such as your wireless card ?
!(https://user-images.githubusercontent.com/36318260/99178997-27422000-273f-11eb-84d4-e3c3918efcac.png)

Zarvis12, 2020-11-15:
> did you enabled options ? such as your wireless card ?

Could it work for this specification
[image]

dubagunta-rohit, 2020-11-26:
@sebanc i would like to try the new kernal..even im facing bluetooth isuues

sebanc, 2020-11-26:
@dubagunta-rohit Sure :) Github email invite sent

gsxin, 2024-02-24:
### Similar issue with me even in 2024

**SOFTWARE USED**
Brunch r121 20240224
ChromeOS hatch R121
Kernel 6.6

**HARDWARE USED**
HP 15s-du1065tu Laptop with rtl8821ce network driver

**IMAGES**

[image]

[image]

**REQUEST**
Please fix it as soon as possible.
bluetoothctl in crosh shell can see and make the other devices pairable but it's not usable with chromeOS.