sebanc/brunch #646: Issues with Keyboard and Bluetooth HP Stream x360 Convertible 11

RiddlemanMVP, 2020-11-14:
Hello Sebanc,

first of all thank you for the great work that gave our HP Stream x360 Convertible 11 new life!

We are running Brunch v86 with Rammus 86 as recovery image.

However, I am facing an issue with the Bluetooth. No Bluetooth devices get recongnised and can be paired. These devices show up normally on our phones. I already read the other thread where some users were complaining of the same behaviour: https://github.com/sebanc/brunch/issues/638

I tried everything that was suggested in the thread but was not able to fix this issue. For completeness I also attach the outcome of the dmseg command. 
[dmesg (1).txt](https://github.com/sebanc/brunch/files/5540390/dmesg.1.txt)
Could you possibly send me the invite to try the latest unstable release?

The second issue I am facing is with the keyboard and the mouse - they are playing up. Sometimes they will work, but most of the times we are stuck in tablet mode and have to type in everything with the touchpad, which is rather annoying. Hardware issue I would exclude, since it works perfectly at boot and also on all the other OS I have tried. 

I have set the flag "chrome://flags/#ash-debug-shortcuts", but this does not seem to reliably enable the shortcut CTRL+SHIFT+ALT+T to switch from tablet to desktop mode. Sometimes it will work, but most of the times not.

I do realise that this might be related to ChromeOS itself, but maybe you had experience with the issue and you have a suggestion.

Thanks

sebanc, 2020-11-14:
Hello,

I am currently extending the tests on 5.4 kernel which seems to improve bluetooth issues. I have just sent you an invite to "brunch-testing" repo which has a 5.4 kernel build, could you try it ?

(A little reminder just in case, phones cannot be paired in chromeos as there is a dedicated menu)

Regarding, the input devices issue, if it persists in 5.4 kernel, I would need some claification on why you think it is related to tablet mode.

RiddlemanMVP, 2020-11-15:
Good Morning Sebanc,

I have tried the testing repository yesterday. Indeed, the Bluetooth situation seems to be better. We were able to connect our devices and stream music to them. Thank you.
However, ChromeOS is still not able to recognise the Bluetooth MIDI service of our digital piano. Do you have any experience with this kind of issue? When using our phones/notebooks, we are able to connect without any issue to the piano.

The keyboard issue still persists. The funny thing is that the keyboard is obviously being recognised by ChromeOS, since I am able to regulate brightness and volume using the foreseen function keys (F2/F3 - F6/F7/F8). This is why I reckon that it's stuck in the tablet mode.

sebanc, 2020-11-15:
Hi,

Chromeos bluetooth support is very limited, it mostly supports input devices and audio streaming device, I think your digital piano would probably not work also on a genuine chromebook.

Regarding the keyboard issue, chromeos automatically maps the keyboard as if it was a chromebook one which is normal and unrelated to tablet mode.

However, your keyboard issue is still very unclear, could you try to clarify it or make a video ? Is it that keypresses are not recognized or letters are assigned to wrong keys...