sebanc/brunch #647: Clear instructions for installing Chrome OS on Legacy/MBR devices

Suleman-Elahi, 2020-11-14:
I tried to install it on my old notebook and used this video as tutorial: https://www.youtube.com/watch?v=UElQAh5SCmo

I used my pendrive as destination to install ChromeOS and now after booting from USB, I am getting this:

[image]

Any idea how to fix this? Thanks.

Suleman-Elahi, 2020-11-14:
Oh... sorry. After having a look at other issues, I got it now... the invalid opcode!!!!

closed 2020-11-14 by Suleman-Elahi

sebanc, 2020-11-14:
Exactly, I am not related to any video tutorial but the youtube should probably mention this which is stated in the readme:

`Supported devices:
Intel hardware (CPU and GPU) starting from 1st generation "Nehalem" (refer to https://en.wikipedia.org/wiki/Intel_Core),`

Suleman-Elahi, 2020-11-14:
Yeah thanks.. I have a i3 based latop... i will install it on that....