sebanc/brunch #648: Touchpad working but not very well (HP ProBook x360 11 G1 EE)

Maxcension, 2020-11-15:
Hello!

I have an HP ProBook x360 11 G1 EE but I have a problem with the touchpad (an Elan one).
The thing is that the touchpad works, but really really bad, like there are some areas of the touchpad that is simply not working (and these areas works great on windows or in the bios).
I tried the ""alt_touchpad_config" / "alt_touchpad_config2"" in the grub config but its but it doesn't change so much.
I'm using the snappy recovery image (because the HP ProBook x360 11 G1 EE also exists with chromeos).

Thanks a lot! 👍

Maxcension, 2020-11-15:
So, i tried some things like "psmouse.elantech_smbus=1" refered in #623, use "psmouse proto=imps" thing or creating file /etc/modprobe.d/elan.conf with "elantech_smbus=0 synaptics_intertouch=0" ( https://bbs.archlinux.org/viewtopic.php?pid=1861098#p1861098 ) but none of these really worked.

So yeah i'm pretty stuck, I think i have to modify the /etc/gesture/40-touchpad_cmt.conf but i dont know what to do exactly..

jafer241177, 2020-11-27:
same problem  in lenovo idealpad flex 3  but no solve   for this big problem no update driver in chrome  all out in bios

Maxcension, 2020-11-30:
Yeah, i think we have to wait for someone with the same computer or a close one and who will have better knowledge than us. :(
Personnally i tried everything i could but i didn't succeed. :/

If anyone comes by and has a solution or plans to help us, it would be greatly appreciated! Thank you very much.

klutzyT, 2022-09-06:
i have the same computer (hp probook x360 11 g1 ee) and i used a coral image on r103 and originally the touch pad was just as you had mentioned sticky and just terrible so i used psmouse.elantech_smbus=1 kernel command line and it fixed itself but now the speakers dont work on here does anyone have a fix for that?

closed 2026-04-01 by github-actions[bot]