sebanc/brunch #650: Dell XPS 7590 OLED

shad0wfile, 2020-11-15:
Hey - we had talked maybe about a year ago re: trying to get oled brightness to work. I've hunted for solutions with no luck yet.

I've seen in some recent posts you have a test build on 5.4 kernel, and I've seen some launchpad articles where they say it may be addressed in that kernel rel.

Any chance for an invite to give it a try?

Thanks!

sebanc, 2020-11-15:
Sure, invite sent :) Try the release in the "brunch-testing" repository.

shad0wfile, 2020-11-15:
Gave it a try, no luck.. back to the waiting game :)

Thanks!

closed 2020-11-15 by shad0wfile

sebanc, 2020-11-23:
Sorry but I have never really understood the problem with OLED, is it that brightness is always at 100% ?

shad0wfile, 2020-11-23:
Yeah, always 100% - and on an OLED.. it's brutal, retina burning in low-lit areas. There has to be some kind of resolution out there, the Samsung Galaxy Chromebook is OLED and they have it working on there, guessing it has some flag somewhere set for it.

sebanc, 2020-11-23:
As far as I can tell, OLED are not supported in the standard linux kernel but is compensated in linux by software color profiles.

Apparently the corresponding way to do this in chromeos would be to set night light to the expected percentage. Could you try it ?

shad0wfile, 2020-11-24:
I gave night light a try, it still stays very bright just in different tones. Yeah I figured as much re: the other chromebook. 

Thanks for the tips and taking a look, I'll keep looking around to see if I can find anything that works.

sebanc, 2020-12-31:
Could you try the release in brunch-unstable repo ?

shad0wfile, 2020-12-31:
I gave it a shot, have it loaded up right now - no luck though. Using the keyboard buttons or trying the manual slider in the chromeos menu just slides side-to-side with brightness staying the same.

Let me know if you want me to collect any additional details on it.

sebanc, 2020-12-31:
Could you check if your brightness work on a ubuntu 20.10 live usb flashdrive ? (I implemented a patch that was present in the ubuntu kernel, maybe your device does not work with it)

shad0wfile, 2020-12-31:
Ah, that may be why, I'll give it a shot, historically the only Linux OS "out of the box" to work is PopOS! (ubuntu derivative) because they sell OLED laptops.

sebanc, 2020-12-31:
I am looking at popos kernel source but I don't find much in it. Could you try to add "i915.enable_dpcd_backlight=1" to the kernel command line ?

shad0wfile, 2020-12-31:
I just tried default ubuntu and it didn't work (verified). Let me try the dpcd backlight command into the latest test rel you pushed, will report back.

shad0wfile, 2020-12-31:
I tossed this in and rebooted:
`linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug i915.enable_dpcd_backlight=1 \`

Didn't change anything, just wanted to verify that's the correct spot for it in the grub conf.

sebanc, 2020-12-31:
Yes that's correct. I will try to apply the "linux-oled" patches from arch aur when they get ported to 5.10

shad0wfile, 2020-12-31:
Sounds good!

sebanc, 2021-01-17:
I applied the linux-oled patches to 5.10 kernel in the latest brunch-unstable build if you want to try it

shad0wfile, 2021-01-17:
I gave it a shot, no luck. I have no idea why this panel won't honor the latest kernel release, much like I have no idea what PopOS could be doing to make it work, very strange.

Thanks for giving it a shot.

sebanc, 2021-01-17:
Did you change "/kernel" to "/kernel-5.10" in the grub config ? (default kernel is 5.4)

shad0wfile, 2021-01-17:
Awesome! That fixed it.. working flawlessly now.

Thanks for all the help!

sebanc, 2021-01-17:
Nice to see that it eventually worked :)

sebanc, 2021-02-07:
Fyi, the OLED brightness patch introduced a regression for some users, starting from the next release you will have to enable it using "options=oled_display"