sebanc/brunch #651: Grub Issue

rob151515, 2020-11-16:
I have followed the instructions and installed to a separate partition on my Surface Go (64gb) model. I am dual booting ChromeOS with WIndows 10. All seems to be working fine except when I select the ChromeOS entry in the Grub menu when booting.

It briefly flashes a black window which says `Booting ChromeOS` then returns me back to the Grub menu. The weird thing is that now there is an extra entry in my Grub menu at the bottom which is titled "ChromeOS (boot from disk image)".  If I then select that entry it boots into ChromeOS and everything is working fine.

Does anyone know how to fix?

Here is the custom code the installer gave me to paste into Grub2win:

```
menuentry "ChromeOS (boot from disk image)" {
        img_part=/dev/mmcblk0p4
        img_path=/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
```

Another thing I noticed was that if I let it sit at the Grub menu and auto select the ChromeOS option (after the timeout), it out puts a message:

```
Booting `ChromeOS

GNU Grub is now loading ChromeOS

  Failed to boot both default and fallback entries.

Press any key to continue...
```

sebanc, 2020-11-20:
Your grub configuration looks fine, I guess your issue is related to grub2win but I don't personally use it so I hope someone else can help with it.

rob151515, 2020-11-20:
Ok no dramas I'll take it up with the grub2win guys 👍

sebanc, 2020-11-20:
Sorry I meant the issue is probably related to the brunch grub2win tutorial and hope another brunch user with grub2win skills will help.

Feel free to re-open the issue.

You can also try asking the grub2win guys but I am not sure they will be interested to help with this

rob151515, 2020-11-20:
Thanks, I'll reopen in the off chance someone else can comment. If I get anywhere with the grub2win guys I'll update here.

brkeejp, 2020-12-10:
you can bypass grub u know? just add EFI entry to your UEFI.
after that you can select chrome os from your UEFI boot oprions

sreefu, 2021-07-23:
@rob151515 Did you ever manage to get this solved? I am having this exact issue with a triple boot setup (Windows 10, Linux Mint Cinnamon & Chrome OS). Added the Chrome OS menu entry from the end of the installer to Grub via Linux Mint.

rob151515, 2021-07-31:
No sorry, was never able to work it out. I ended up deleting Windows and just using Chrome OS. Windows is terrible on a base model Surface Go anyway.

anuragtambe200, 2021-08-05:
Disabling Secure Boot From the BIOS might help. Doing this is different for every motherboard, so you might have to look it up.

sreefu, 2021-08-05:
> No sorry, was never able to work it out. I ended up deleting Windows and just using Chrome OS. Windows is terrible on a base model Surface Go anyway.

I managed to solve this by removing:
`menuentry "ChromeOS (boot from disk image)" {`

from the Grub and just started with it's parameters:
`img_part=/dev/mmcblk0p4` and etc. 

Don't forget to remove the closing parentheses at the end too. Once thats done, selecting Chrome OS boots right into Brunch and loads Chrome OS perfectly without creating any loop entries.