sebanc/brunch #654: Two problems

wggglggg, 2020-11-17:
1 .  Can`t open any APK,or Google app store , when reboot or boot login system, all the apps are circling, I found a way to let they work , first you click one app, then logout ,then login again, then click one app , obviously it`s circling again, then just wait for 1min or 2min  apk works .
[https://ibb.co/mCLhpsv](url)

2.  Sometimes can`t watch video on Web using chrome or Firefox  ect. YouTube Youke Iqiyi any video can`t watch , YouTube app either. sometimes all works, I don`t know, I didn`t do anything just reboot,   failed times more then working times. 
[https://ibb.co/9q52YPX](url)

my pc is ThinkPad edge S420 ,    I5 2450, 6G ddr3, inter and HD
[https://ibb.co/zGx9jS3](url)

sebanc, 2020-11-17:
@wggglggg You should have received a github invite to "brunch-testing" repo, could you try the release with 5.4 kernel there ?

wggglggg, 2020-11-17:
> @wggglggg You should have received a github invite to "brunch-testing" repo, could you try the release with 5.4 kernel there ?

and i found a funny thing about the No. 2 problem,   
1.   boot my PC, and login the Chrome OS system,  then try to let the app( V2ray ) go, one of apps I`m using because i`m in China .  
2.   when the app  works , open Chrome web to Youtube.com click any video ,  just wait it few minutes. if can watch the video, so can watch any web videos. 
I tried many time, it seems the way can make it  .   try 10 time,  works 8 time  or more.
Does that mean, chrome needs some  plugs from Google, when first time use it? whatever you halt your PC, you should go again.
very Interesting.
anyway,  i`ll try the test brunch_r86_k5.4_unstable_20201114.tar.gz. later

closed 2020-11-18 by wggglggg