sebanc/brunch #655: k5.4 unstable 20201114 Google App Store can`t install- ThinkPad edge S420

wggglggg, 2020-11-17:
5.4 unstable  +  chromeos_13421.73.0_samus_recovery_stable-channel_mp-v3.bin,   system show there`s something wrong about Google App Store, following the picture, can`t open it.

https://ibb.co/tJhc1Rc

and , try another club ( 5.4 unstable + chromeos_13310.59.0_samus_recovery_stable-channel_mp-v3.bin ).  app store  is working now.

sebanc, 2020-11-17:
A google search shows that your laptop has an AMD gpu, could you try disabling it in the bios (if possible) ?

If you cannot disable the AMD gpu in the bios, please attach an lsmod to this issue (run `lsmod > ~/Downloads/lsmod.txt` in crosh shell and attach lsmod.txt from your Downloads folder.

wggglggg, 2020-11-17:
Do u mean type RUN LSMOD in system? Where to type? Which app? And the whole code is?

ameyjoshi3068, 2020-11-17:
@wggglggg here are the steps
1. press ctrl+alt+t this will open crosh terminal.
2. Type `shell` and press enter.
3. Run `lsmod > ~/Downloads/lsmod.txt`

wggglggg, 2020-11-17:
really helpful,  I'll do it today later, thank you

发自我的小米手机

wggglggg, 2020-11-18:
bash: run : command not found

发自我的小米手机

wggglggg, 2020-11-18:
I`m sure about watching Youku  Iqiyi Tencent Video on WEB,
you must open v2ray  or the same kind of apps ,before you watch them on website, chrome os  needs download something through VPN just few seconds ( 5~10 sec). Unless you reboot pc  or log out system, then you can watch them even leave VPN away,

wggglggg, 2020-11-18:
> A google search shows that your laptop has an AMD gpu, could you try disabling it in the bios (if possible) ?
> 
> If you cannot disable the AMD gpu in the bios, please attach an lsmod to this issue (run `lsmod > ~/Downloads/lsmod.txt` in crosh shell and attach lsmod.txt from your Downloads folder).

I`ve checked the BIOS, it`s Integrated Graphics already, 
pictures following:    
https://ibb.co/gjk9p7h
https://ibb.co/tJhc1Rc

wggglggg, 2020-11-18:
chronos@localhost / $ lsmod
Module                  Size  Used by
x86_pkg_temp_thermal    16384  0
intel_powerclamp       16384  0
crc32c_intel           24576  0
ghash_clmulni_intel    16384  0
aesni_intel           368640  21700
crypto_simd            16384  1 aesni_intel
glue_helper            16384  1 aesni_intel
psmouse               135168  0
intel_rapl_msr         16384  0
thinkpad_acpi          90112  0
hid_logitech_hidpp     32768  0
hid_logitech_dj        24576  0
snd_hda_codec_hdmi     49152  1
snd_hda_codec_conexant    20480  1
snd_hda_codec_generic    69632  1 snd_hda_codec_conexant
ledtrig_audio          16384  3 snd_hda_codec_generic,snd_hda_codec_conexant,thinkpad_acpi
snd_hda_intel          36864  3
rtl8192ce              57344  0
snd_intel_dspcfg       24576  1 snd_hda_intel
rtl_pci                32768  1 rtl8192ce
snd_hda_codec         110592  4 snd_hda_codec_generic,snd_hda_codec_conexant,snd_hda_codec_hdmi,snd_hda_intel
rtl8192c_common        45056  1 rtl8192ce
rtlwifi                98304  3 rtl_pci,rtl8192c_common,rtl8192ce
snd_hda_core           69632  5 snd_hda_codec_generic,snd_hda_codec_conexant,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec
r8169                  77824  0
realtek                20480  1
snd_hwdep              16384  1 snd_hda_codec
mei_hdcp               16384  0
libphy                 65536  2 r8169,realtek
mei_me                 24576  1
mei                    69632  3 mei_hdcp,mei_me
i915                 1708032  34
intel_gtt              20480  1 i915
drm_kms_helper        135168  1 i915

sebanc, 2020-11-18:
Indeed, it is integrated graphics, I can see only three things which could cause that:
- An incorrect Google sync, try resetting it: Login to your Google dashboard account -> Go to the "Chrome" application -> Manage synced data and click on "reinitialize synced data"
- The use of a slow usb flashdrive
- Your VPN

wggglggg, 2020-11-18:
Nothing wrong with VPN , and the OS is loaded in the harddisk,  so I tried to reinitialize synced data,     then The Web Problem is done. 
The apps circling problem has been salved, even though it`s not a good choices, like I mentioned  log in the system,  click any app , off course it`s circling , just leave them there,  log out the OS ( not reboot ), log in again, click app again, wait in   2min , the app start. 
Anyway, thank you for your suggesting.

wggglggg, 2020-11-18:
> Indeed, it is integrated graphics, I can see only three things which could cause that:
>
> * An incorrect Google sync, try resetting it: Login to your Google dashboard account -> Go to the "Chrome" [...]

Nothing wrong with VPN , and the OS is loaded in the harddisk, so I tried to reinitialize synced data, then The Web Problem is done.
The apps circling problem has been salved, even though its not a good choices, like I mentioned log in the system, click any app , off course its circling , just leave them there, log out the OS ( not reboot ), log in again, click app again, wait in 2min , the app start.
Anyway, thank you for your suggesting.

closed 2020-11-18 by wggglggg