sebanc/brunch #657: Microphone Issue

yegender124001, 2020-11-18:
I have a desktop and installed Chrome OS on it. I have a headphone of SAMSUNG. It is not showing in sound menu but voice is properly comming. I want to use it's microphone for my classes on Google Meet. It's microphone is not working. How can I enable it's microphone. It can be used for mic in android but not in my chrome desktop. Please Help Me

Thanks

sebanc, 2020-11-20:
You might want to try "internal_mic_fix" option and also to look at the issue #56

klogod, 2020-12-05:
Excuse me, where should I put this command line?

> 
> 
> You might want to try "internal_mic_fix" option and also to look at the issue #56

sebanc, 2020-12-05:
refer to https://github.com/sebanc/brunch#framework-options

or #603

klogod, 2020-12-05:
> refer to https://github.com/sebanc/brunch#framework-options
> 
> or #603

I tried to add "options=internal_mic_fix" but no lock, its that mean my internal microphone cannot be enable?

neerajverma04, 2021-04-29:
this command works for me in shell amixer -c0 set "Internal Mic" toggle

If it gives Invalid card number... then try c1

closed 2026-04-01 by github-actions[bot]