sebanc/brunch #659: DELL Inspiron 1545, boot problems with ChromeOS

Raffaco, 2020-11-20:
Hi, i have a problem with the ChromeOS installation.
I followed all the instructions from the youtube page "https://www.youtube.com/watch?v=ROBpXNonVxc".
Installation did go well without any errors, however when i try to boot the ChromeOS it keeps at this screen forever (see picture).
I also tried diffrent iso's , none of them seems to work.
Am i doing something wrong? Please help me. 

**Here is what i did:**
* Downloaded Linux Mint.iso
* Used Rufus to flash usb drive
* Downloaded Brunch  	        (version: r86 k4.19 stable 20201014)
* Install.Sh  		                (from: https://raw.githubusercontent.com/shrikant2002/ChromeOS/master/install.sh)
* Downloaded Chrome OS 	
(13421.73.0_rammus_recovery_stable)
(13421.73.0_samus_recovery_stable)

**My laptop specifications:**
* DELL Inspiron 1545
* Intel Pentium Dual Core T4200 CPU
* Intel GMA 4500MHD GPU
* 2GB DDR2 ram
* 160GB HDD

[image]

sebanc, 2020-11-20:
One of the problems with those videos is that they don't mention compatibility aspects and notably that brunch is not compatible with core 2 duo

paregl, 2021-12-29:
did you got it running? chrome os on intel t4200 processor? please share your experience. thx

closed 2026-04-01 by github-actions[bot]