sebanc/brunch #662: surface pro 4 touchscreen randomly stops working

hassanhabbak, 2020-11-20:
I am on rammus 86 image with brunch version r86 k4.19. All works great except that touchscreen stops responding to touch randomly every 10minutes or so. The OS continue to respond to mouse and keyboard though. Any ideas where to start looking for the issue?

sebanc, 2020-11-20:
Hi,

Some SP4/SB1 need a special kernel parameter for touchscreen not to fail. Could you try adding "intel-ipts.no_feedback=1" to the kernel command line ?

hassanhabbak, 2020-11-21:
That did the trick! Happy to report everything works great including internal sensors. The only thing not working are the cameras. Many thanks for a fantastic project!

jjohns829, 2020-11-25:
Hello all!

I am a noob and tried to search for the process to accomplish the kernel command line edit, bit to no avail.  Is is through Linux terminal, Crosh, something.  Please advise and thank you in advance.  Also happy holidays!

sebanc, 2020-11-26:
Hi @jjohns829, you can refer to #603

jjohns829, 2020-11-26:
Thanks @sebanc

jjohns829, 2020-12-05:
Hi!

Updates Bruch framework to: r86 k5.4 stable 20201201 and touch screen on
Surface Pro 4 does not function at all.  Added "intel-ipts.no_feedback=1"
to the kernel command line, but to no avail.

Best,
Matt

jjohns829, 2020-12-05:
Updates Bruch framework to: r86 k5.4 stable 20201201 and touch screen on Surface Pro 4 does not function at all.  Added "intel-ipts.no_feedback=1" to the kernel command line, but to no avail.

sebanc, 2020-12-05:
With kernel 5.4, you need to add "options=ipts" on the kernel command line in your grub configuration (after cros_debug)

jjohns829, 2020-12-05:
Thanks @sebanc!  Would I still keep the "intel-ipts.no_feedback=1" kernel parameter?

sebanc, 2020-12-06:
The "intel-ipts.no_feedback=1" is not needed anymore