sebanc/brunch #664: HP Elitebook 1040 G5 - Battery drain when in sleep mode

ashnazir786, 2020-11-22:
Hi,

Thanks for the great work. 

I only have one problem and that is when the laptop goes into sleep mode I have a fair amount of battery drain. If I shut it down completely I do not have this problem, just when in sleep mode. Also, I can also not wake it from sleep mode by simply pressing the power button. I need to press and hold for at least 5 seconds. Not sure if this points to what the problem might be.

Thanks
Ash.

sebanc, 2020-11-23:
Hi,

You should have received a github email invite to the "brunch-testing" repo which contains a brunch build with kernel 5.4. Could you try to see if it helps with your issue ?

ashnazir786, 2020-11-23:
Thank you I've got the email. Do I have to totally wipe the laptop or is it
possible to apply this without removing everything ?

Thanks
Ashiq

sebanc, 2020-11-23:
Updating should work but there is always a risk that it won't boot and you will loose your data

ashnazir786, 2020-11-23:
Can you give me the link to the updating procedure ? I've only done this as
a new install.  Thanks.

biogon88, 2022-01-18:
Was this fix integrated into current builds? Having a similar experience on Surface Go 2 -- been on standby from 11am to 6pm (7 hours) and the battery has drained from 100% to 62%. 

Thanks!