sebanc/brunch #666: Brightness low on NOCTURNE(Pixel Slate)

usagicore, 2020-11-23:
Brightness set on lowest, changes only off and lowest.
/sys/class/backlight/intel_backlight/max_brightness is 7500. (above 20000 on stock firmware)
Manually changing /sys/class/backlight/intel_backlight/brightness does not work.

Kernel parameters tried : (dev.)i915.enable_dpcd_backlight

sebanc, 2020-11-24:
You should have received a github email invite to "brunch-testing" repository, could you try the build with 5.4 kernel there ?

usagicore, 2020-11-25:
It works very well with 5.4 kernel. Dual boot works well too.

usagicore, 2020-11-29:
Update : Sound / fingerprint does not seem to be working.

olm3ca, 2022-11-29:
@YounsooChoi did you ever discover a solution for sound? I'm trying brunch on the Pixel Slate and have the same problem.

olm3ca, 2023-01-19:
For anyone looking for a fix to sound: go to /etc/modprobe.d and add avs.conf with the following line: `options snd-intel-dspcfg dsp_driver=4`