sebanc/brunch #669: Black screen after Booting Log show something call 'EFI vga' on  Dell Inspiron 3162/3164

klogod, 2020-11-24:
When I select to Boot Chrome OS from the grub menu, everything is loading fine, but the screen just go black after something call 'XXXXXEFI vga'. 
It's too fast so i cant read the whole log.
Grub menu is normal.

I installed the r86 version with rammus recovery.

sebanc, 2020-11-24:
You should have received a github email invite to "brunch-testing" repository, could you try the build with 5.4 kernel there ?

klogod, 2020-11-25:
> You should have received a github email invite to "brunch-testing" repository, could you try the build with 5.4 kernel there ?

OK, I will try.

mirek190, 2020-11-26:
Hello sebanc .
Can you add me to brunch-testing as well ? 
I also want to make tests with new kernel 5.x line .

sebanc, 2020-11-26:
@mirek190 Github email invite sent.

mirek190, 2020-11-26:
Thanks sebanc

klogod, 2020-11-26:
Its's work now, thanks sebanc

xp1ode, 2020-11-27:
anyway to get an invite to the brunch-testing as well?

sebanc, 2020-11-27:
@xp1ode Sure, Github email invite sent.

Bigwil63, 2020-11-30:
Having the same issue on Microsoft Surface 3.  Everything great and then a normal reboot caused the "...EFI vga" in the boot log.  It does boot, but is painfully slow and unusable (never made it past the login).  Tried to boot from the original USB and same thing.  Maybe the 5.4 kernel would help me too.

klogod, 2020-11-30:
Try to disable all about "legacy ROM" in BIOS.

> Having the same issue on Microsoft Surface 3. Everything great and then a normal reboot caused the "...EFI vga" in the boot log. It does boot, but is painfully slow and unusable (never made it past [...]

Bigwil63, 2020-11-30:
I don't have that option on the surface 3, I am familiar with it on other PCs.  TPM and Secure Boot Control are both disabled though.  Experimented with turning them on and back off with no change.  Thanks

sebanc, 2020-11-30:
@Bigwil63 you should have received a brunch-testing github email invite.

Pentahex, 2020-12-02:
I would also request a brunch-testing invite, love this project, I have it installed on three of my laptops.

stateandrei5, 2020-12-02:
Can i get one too?

sebanc, 2020-12-02:
@stateandrei5 the newest release is now in the main brunch repo

stateandrei5, 2020-12-02:
> @stateandrei5 the newest release is now in the main brunch repo

yep. im clean installing. currently out for a smoke while waiting on framework rebuild