sebanc/brunch #670: compatibility with Pentium N3700

klogod, 2020-11-24:
Is brunch compatibility with Pentium N3700?

sar777-dev, 2020-11-24:
I had already experimented on even low performance processor J2900  which more weaker than yours and it worked.
You should give it a try, using live boot usb, if it performs well then you can install it if want.

klogod, 2020-11-24:
> 
> 
> I had already experimented on even low performance processor J2900 which more weaker than yours and it worked.
> You should give it a try, using live boot usb, if it performs well then you can install it if want.

You mean install the Chrome OS on USB?
I've tried rammus, samus and grunt install on SSD with no luck...

sebanc, 2020-11-24:
In theory both rammus and samus recovery images should work.

What's your laptop model ? and what happens when you try to boot brunch ? (black screen, some text is displayed on the screen then it reboots, you see the chromeos logo but it does not go further)