sebanc/brunch #671: Issue with play store.

darkxduck, 2020-11-27:
Hi, first of all this project is amazing and very interesting, thank you!
I have a problem with the play store, It takes an eternity to open and sometimes i just doesn't and sometimes it opens instantly, its very inconsistent and i want to fix that. I try with the play store fix config but nothing, it's the same.

Thanks in advance.

sebanc, 2020-11-27:
You should have received a github email invite to "brunch-testing" repository, could you try the build there with 5.4 kernel ?

darkxduck, 2020-11-28:
Yes thanks ill try but i think its related to the issue of the play store doesn't open unless you wait 2 mins in the login screen.

sar777-dev, 2020-11-29:
@sebanc  
Hi sebanc, I wanted to know that this "brunch-testing" have some performance tweaks than 4.19 kernal, if that's true then can you also give this "brunch-testing" with 5.4 kernal to me.

sebanc, 2020-11-29:
@sar777-dev Not sure it will change much in terms of performance, I mostly tried to fix the play store issue. In any case, you should have received the invite to "brunch-testing".

closed 2026-03-31 by github-actions[bot]