sebanc/brunch #672: Asus UX303LA (UX303LAB) Touchpad Not Working

metalzero24, 2020-11-27:
Hi,
I installed chromeos using Brunch r86 k4.19 stable 20201014 (latest). My touchpad is not working. Here are the outputs for "udevadm info" mouse0 and mouse1" I attached "udevadm info" outputs for mouse0 and mouse1.

[image]
[image]

Laptop: Asus UX303LA (UX303LAB)

darkxduck, 2020-11-27:
try the touchpad fixes they worked for me

metalzero24, 2020-11-27:
> try the touchpad fixes they worked for me

I went to terminal as root and tried options=alt_touchpad config and config2
then typed reboot
same problem nothing works

closed 2023-02-18 by selimsandal