sebanc/brunch #675: RTL8822CE bluetooth working, no wifi

hartleyshc, 2020-11-29:
Using brunch v86
Grunt recovery 86 (A6 processor, laptop)

Read through previous issues, have tried booting with options=rtl8822 
Read through hardware options, don't see any options for rtl8822ce. Not seeing any beta (I assume beta = brunch-testing) versions for 86. 

Can't ever get WiFi to be enabled, seems from other posts users have either WiFi working, or BT+ WiFi working.

sebanc, 2020-11-30:
rtl8822ce should work out of the box so I don't think that it's actually your wifi driver, what's your laptop model ?

Da-Jayman, 2020-12-04:
Did you figure this out? I have the same issue. I was able to load the driver for Linux but cant get it to work with Chrome OS 86

hartleyshc, 2020-12-04:
@sebanc lenovo ideapad slim 1-14ast-05. I'm not sure if this model offers other cards, but I can confirm by physically looking at the card, plus having to install the rtw88 driver in various Linux distros that it is the RTL8822CE. 

@Da-Jayman I swapped it out with an Intel card I have from another laptop. But it didn't seem too stable with the iwlwifi driver, so I swapped it back with the rtl8822ce, and now I'm using my old tried and true ralink rt5370. It's an old, cheap, usb based WiFi-N plug, that's worked with anything I've thrown at it for about a decade. (Support was added to the Linux kernel maybe 5-6 years ago).

sebanc, 2020-12-04:
This is weird, I just verified that the driver is correctly enabled in the kernel.

Are you both using the latest 5.4 release ?

sebanc, 2020-12-04:
> @sebanc lenovo ideapad slim 1-14ast-05. I'm not sure if this model offers other cards, but I can confirm by physically looking at the card, plus having to install the rtw88 driver in various Linux distros that it is the RTL8822CE.
> 

In linux do you still need the rtw88 driver on distros using kernel 5.4 ?

hartleyshc, 2020-12-04:
I was using the r86 with 4.19 kernel.

This is actually a new laptop ($100 special, planned on using it as mostly a terminal device). 

I'll try latest release and see if updated kernel helps. If I get the same issue, I'll install some Debian based system and load up kernel 5.4 and see what happens with that.

Da-Jayman, 2020-12-04:
So I went into Mint and found out my wifi card was rtl8821ce
I entered  options=rtl8821ce in the grub menu and it worked !!!

SK82DEC, 2021-06-25:
> Using brunch v86
> Grunt recovery 86 (A6 processor, laptop)
> 
> Read through previous issues, have tried booting with options=rtl8822 also tried options=rtw88 which is the name of the driver realtek contributed to the Linux kernel.
> Read through hardware options, don't see any options for rtl8822ce. Not seeing any beta (I assume beta = brunch-testing) versions for 86. (edit: I see that brunch-testing is a private channel)
> Can't ever get WiFi to be enabled, seems from other posts users have either WiFi working, or BT+ WiFi working. For me BT is working, but WiFi is not.

I own the same machine and struck in the same problem. Did you find any solution?

Da-Jayman, 2021-06-28:
I just entered options=rtl8821ce in the grub menu and it worked !!! It was a while ago so I dont remember all the steps but that got the wifi working.