sebanc/brunch #677: Backup and migrate Chrome OS

stucki, 2020-11-30:
This is a follow-up to #298 which was closed without a solution.

My Brunch setup is run from an image file on the local disk. I'm using btrfs for this which allows creating snapshots (https://github.com/sebanc/brunch/issues/512#issuecomment-695947846). So far this is working almost perfect.

I've been trying to copy my system image file to another computer and boot the same system from there. This boots fine, however all data is gone and I need to configure everything from scratch again.

I would like to know why the existing data is ignored / removed and if this can be provented.

Scenarios where this could be useful:
- Take your Chrome OS on a memory stick wherever you go
- Create backups of just the data partition (which is much smaller compared to a full image)

Thanks in advance for any suggestions!

stucki, 2020-12-10:
Any ideas? This would be a great feature.

I could try to investigate this myself. However, where to start? Thanks for any suggestions...

sebanc, 2020-12-11:
This is most likely a chromeos security feature but I have not yet seen in the code where this process is located. I guess that it checks for DMI values or something like that and prevents mounting the encrypted storage if they are different than the stored one.

To be honest, I consider this as a valuable security feature and I don't really intend to work on this.

stucki, 2020-12-11:
Thanks for the feedback!