sebanc/brunch #678: Hp Elitebook 850 G1 i5 4300U

fctaddia, 2020-12-01:
Hi, I have installed Chrome OS r83 on my hp 850 g1 and everything works but it often jerks and the pc is almost always too hot. I was wondering if my pc was compatible and or if I had to install a different version of Chrome OS.

sebanc, 2020-12-02:
Hi, it seems to be related to the amd gpu, try adding "module_blacklist=amdgpu,radeon" to the grub configuration (on the kernel command line -> after cros_debug)

fctaddia, 2020-12-02:
I have the Intel Graphics 4400

sebanc, 2020-12-02:
You don't have an amd dGPU (in addition to intel graphics) ? the specs looked like your computer had one

fctaddia, 2020-12-02:
There are different versions this and the version of my cpu in which there is also the graphics

https://ark.intel.com/content/www/it/it/ark/products/76308/intel-core-i5-4300u-processor-3m-cache-up-to-2-90-ghz.html