sebanc/brunch #679: Boot Error

Adrianchot3, 2020-12-02:
Hello, I am using the google translator sorry, I have a problem, after executing the commands in Linux mint, I restart my PC and when starting Chrome OS, I have the error that is seen in the image.

My PC:
Intel Pentium Dual Core E5200
2 GB RAM
BIOS (NO UEFI SUPPORT)
[image]

sar777-dev, 2020-12-02:
If I remember correctly Pentium dual core is not supported yet on brunch ChromeOS.

Adrianchot3, 2020-12-02:
> If I remember correctly Pentium dual core is not supported yet on brunch ChromeOS.

When will brunch be available for dual core pentium processors? And ... the Intel Celeron 847 processor supports brunch?

sebanc, 2020-12-02:
Chromeos recovery images are not compatible with core 2 duo, unfortunately I cannot do anything about that

Adrianchot3, 2020-12-02:
> Chromeos recovery images are not compatible with core 2 duo, unfortunately I cannot do anything about that

And ... the Intel Celeron 847 processor supports brunch?

sebanc, 2020-12-02:
I don't remember tbh but only cpu with sse4_2 are supported and some old Celeron have it

Adrianchot3, 2020-12-02:
> I don't remember tbh but only cpu with "sse4_2" cpu instructions are supported and some old Celeron have it

On the official Intel page it shows me what is in the attached image, and it seems that the intel celeron 847 is compatible or am I wrong?

Adrianchot3, 2020-12-02:
> I don't remember tbh but only cpu with "sse4_2" cpu instructions are supported and some old Celeron have it

On the official Intel page it shows me what is in the attached image, and it seems that the intel celeron 847 is compatible or am I wrong?
[image]

sebanc, 2020-12-02:
If it has "sse4_2" it should work

Adrianchot3, 2020-12-02:
> Si tiene "sse4_2" debería funcionar

Well in the picture it shows that it has sse4.1 and sse4.2 so it must be compatible, thank you, I hope your project is successful!

Adrianchot3, 2020-12-06:
> > Si tiene "sse4_2" debería funcionar
> 
> Well in the picture it shows that it has sse4.1 and sse4.2 so it must be compatible, thank you, I hope your project is successful!

Hola otra vez, Chrome os funciona correctamente en un intel Celeron 847!