sebanc/brunch #680: Bootloop on Atom Z3745 Tablet

ludaka, 2020-12-03:
Hey i have this issue on my Intel Atom Z3745 Tablet with x32 bit UEFI Bios
p.s. As i seen from other ppl who have tablets sticks with same processor they don't have such issue .. is it related on last brunch or need to check something in UEFI ?

[image]

sebanc, 2020-12-04:
Hi, I had a previous report for the same issue from a user on telegram but he did not try to debug it.

You should have received a github email invite for the brunch-testing repo, could you try the release there ?

Prazaar, 2020-12-16:
error also with intel celeron n2807, new brunch with old kernel works well

sebanc, 2020-12-17:
Could you try the new "brunch-unstable" build ?

Prazaar, 2020-12-21:
brunch-unstable boots without any problems

sebanc, 2020-12-21:
Just to confirm brunch-unstable is able to boot with kernel 5.4 ?

Prazaar, 2020-12-21:
correctly.
i update from stable to unstable without any problems, kernel is 5.4.