sebanc/brunch #683: Two issues on Dell Latitude E5470

PXNDA1234321, 2020-12-04:
1. Error booting up
    `error: no module found...` (or something similar). Doesn't really affect the machine as far as I know. I just press a key and it boots up normally.

2. INTERNAL MIC NOT WORKING
    This is a bummer for me. I got my mic working on Arch Linux by installing it pulseaudio and I am not sure if I could do something similar on chromeOS. I've looked at other issues but still haven't been able to fix it.

Thanks in advance and thank you sebanc for this awesome project.

ameyjoshi3068, 2020-12-05:
Hi,
Your first issue seems to be related to grub entry. removing `rmmod tpm` from the first line should fix this issue.

sebanc, 2020-12-05:
Hi,

ameyjoshi3068 solution is correct for 1.

For 2. it seems the first 5.4 kernel build introduced a regression wrt microphone, I have sent you a github email invite to the "brunch-testing" repo, could you validate it and try the build there ?

If microphone still does not work, could you try to add a space and "options=internal_mic_fix" to your grub config after "cros_debug" ?

PXNDA1234321, 2020-12-05:
Hey,

Thanks ameyjoshi3068, that fixed it.

sebanc how should I try the build. Would I use this command: `sudo chromeos-update -f` or would I have to remake the .img?

sebanc, 2020-12-05:
Yes you can use `chromeos-update -f`

PXNDA1234321, 2020-12-05:
Hey sebanc,

Got an error: `bash: chromeos-update: command not found`

sebanc, 2020-12-05:
did you use sudo ?

PXNDA1234321, 2020-12-05:
Oops, just tried it with sudo and got another error: `Brunch release archive not found`

sebanc, 2020-12-05:
You don't need to extract the archive.

run: `sudo chromeos-update -f ~/Downloads/<archive_name>`

PXNDA1234321, 2020-12-05:
Tried the brunch testing repo which, unfortunately, didn't solve the microphone. I am going to try the `options=internal_mic_fix` in grub config.

PXNDA1234321, 2020-12-05:
Nope, no luck sebanc. I've got the `options=internal_mic_fix` after `cros_debug` but it still doesn't work.

PXNDA1234321, 2020-12-07:
@sebanc ?

sebanc, 2020-12-11:
I needed to think about it, here are my 2 last ideas to hopefully fix this:

1)You would need to find your sound card model at:
https://www.kernel.org/doc/html/latest/sound/hd-audio/models.html

And add options snd_hda_intel model=[YOUR_MODEL] to /etc/modprobe/alsa.conf

It works the same on linux, you can also browse the linux forums to find the correct model, an example with a different DELL device (in the first comment):
https://www.linuxquestions.org/questions/linux-laptop-and-netbook-25/internal-microphone-not-working-debian-squeeze-dell-latitude-e6410-854320/

2) Try the "disable_intel_hda" option (chances that it will work are very low)

PXNDA1234321, 2020-12-12:
Holy shit!

I followed sebanc's advice but it didn't work. I didn't want to give up so using my knowledge of Arch I dug around and tried `dmesg | grep snd`. Voila! My codec showed up. So I knew chromeOS was picking it up, but how should I enable it? I then remembered the alsamixer command. I try it and it works. Okay, only one option. winkey + F6 and changed the card to `HDA Intel PCH`. Now we're talking. winkey + F4 to see capture devices and go to internal mic. DISABLED. FRICKING DISABLED! Smash space bar, capture is now on Head over to https://www.onlinemictest.com/ and yes! It works.

I am not closing this yet as I am still somewhat researching this as I have the `internal_mic_fix` option. I'll try to remove it and see if it works.

PXNDA1234321, 2020-12-12:
After removing `internal_mic_fix`, the mic still works but with one issue: You have to change the capture to mic after reboot. This might be a one-off though.

sebanc, 2020-12-13:
What's the output of "aplay -l" ?

Could you post a screen capture of the alsamixer page where you have to unmute the mic ?

PXNDA1234321, 2020-12-13:
`aplay -l`:
card 0: PCH [HDA Intel PCH], device 0: ALC3235 Analog [ALC3235 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

[image]
After reboot.

sebanc, 2020-12-13:
Instead of switching manually the microphone, does `amixer -q -c 0 set "Internal Mic" toggle` has the same effect ?

PXNDA1234321, 2020-12-13:
Yup, `amixer -q -c 0 set "Internal Mic" toggle` does do the same thing.

sebanc, 2020-12-13:
That's exactly the command in "internal_mic_fix" so I don't understand why it does not work... could you post a dmesg while "internal_mic_fix is enabled ? (Run "dmesg > ~/Downloads/dmesg.txt" and post dmesg.txt from your Downloads folder)

PXNDA1234321, 2020-12-13:
[dmesg.txt](https://github.com/sebanc/brunch/files/5684543/dmesg.txt)
Here ya go.

sebanc, 2020-12-13:
options=internal_mic_fix is not present on the kernel command line (3nd line of the log), do you run `sudo update-grub` after modifying your grub config ?

PXNDA1234321, 2020-12-13:
I removed it to see if mic still worked. It did and I didn't add it again. Do I need it and what does it do?

sebanc, 2020-12-15:
No pb, as long as it works, I guess it is ok