sebanc/brunch #685: Chrome OS is unable to boot up with latest 5.4 kernel(clean install)

ameyjoshi3068, 2020-12-05:
I have attached a full video of problem 
https://drive.google.com/file/d/1Or-qhc_WoSGKQufq0zeySEDdU6OH6u_G/view?usp=sharing
the same screen stays on for 1-2 minute and afterwards pc restarts. it keeps happening repeatedly.
I have installed chrome OS, linuxmint, windows 10 on dell inspiron 3647 (ntel(R) Pentium(R) Processor G3240)(4gb RAM)(Intel HD Graphics)

there is also a video of all options in BIOS in case something is wrong there.

ameyjoshi3068, 2020-12-06:
I explored some [arch linux wiki](https://bbs.archlinux.org/viewtopic.php?id=250332) and came to know that blacklisting gsmi can help but there is no blacklist.conf in modprobe.d in root A
I also tried to add module_blacklist=gsmi after cros debug but it didn't worked.

sebanc, 2020-12-06:
Hi, I have looked into your issue but I am not sure yet what could be the cause for that. It seems some devices that booted fine with chromiumos 5.4 kernel r85 (kernel I was using previously) are having issues with chromiumos kernel 5.4 r88. I am going to push a test build with the r85 kernel to confirm if it is that or not. (but probably tomorrow as I am quite busy today).

Regarding gsmi, it is not enabled in my kernel config (the CONFIG_GOOGLE_FIRMWARE dependency is not set)

ameyjoshi3068, 2020-12-06:
Yes probably that can fix the issue.
Thanks you.

ameyjoshi3068, 2020-12-11:
Will it help to have chrome os beta 88 with latest brunch(Because it uses r88 kernel)?
Sorry to bother you

ameyjoshi3068, 2020-12-12:
That sounds great.
Thank you

ameyjoshi3068, 2020-12-12:
Just tried the latest testing build. 4.19 kernel in it just worked flawlessly for me.
It is a very great update. Thanks.

ameyjoshi3068, 2020-12-28:
Is there any chance that 5.4 kernel will work with my pc?

sebanc, 2020-12-28:
It still does not work with latest release ?

ameyjoshi3068, 2020-12-28:
No only 4.19 works

ameyjoshi3068, 2020-12-28:
I tried latest stable release

ameyjoshi3068, 2021-02-01:
video of boot process

https://drive.google.com/file/d/1NhZ8qyLi0XtGTslCTAD7c3vYTRplkkOL/view?usp=sharing

sebanc, 2021-02-01:
Could you try to add "intel_iommu=on" on the kernel command line ?

sebanc, 2021-02-01:
A potential fix was included in the latest brunch unstable, let me know how it goes

ameyjoshi3068, 2021-02-02:
I tried adding `intel_iommu=on` after cros debug and before options. But it did not work. I am trying unstable release now.

ameyjoshi3068, 2021-02-02:
> A potential fix was included in the latest brunch unstable, let me know how it goes

I tried the fix but it didn`t made any difference.

closed 2026-03-31 by github-actions[bot]