sebanc/brunch #686: Dual boot on an BIOS/MBR devices?

glowingnec, 2020-12-06:
Just like the title, is there anyway to dual boot ChromeOS on an old Legacy BIOS system?
I tried the "mbr_support" but it's didn't give me the grub code at the end
Here is my output:
`sudo bash chromeos-install.sh -src /media/nectar/Data/ChromeOS/samus.bin -dst ~/tmpmount/chromeos.img -s 25
Creating image file
0+0 records in
0+0 records out
0 bytes copied, 0,000579884 s, 0,0 kB/s
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
7673DF71-A813-4643-92AB-2DDB6BC5AD65
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
The operation has completed successfully.
       start        size    part  contents
           0           1          PMBR (Boot GUID: 7673DF71-A813-4643-92AB-2DDB6BC5AD65)
           1           1          Pri GPT header
           2          32          Pri GPT table
    19255304    33173448       1  Label: "STATE"
                                  Type: Linux data
                                  UUID: A1A1547F-8012-494E-A12A-AF36C7193660
     2117638       65536       2  Label: "KERN-A"
                                  Type: ChromeOS kernel
                                  UUID: CCBDB2EF-34EE-2749-946F-5DFE2C23B8C5
                                  Attr: priority=15 tries=15 successful=0 
    10866696     8388608       3  Label: "ROOT-A"
                                  Type: ChromeOS rootfs
                                  UUID: 4A7A9BB9-7B74-D244-B999-3CA375F1842F
     2183174       65536       4  Label: "KERN-B"
                                  Type: ChromeOS kernel
                                  UUID: 7FC79F76-1B5D-7B41-BD10-7083CABFE01C
                                  Attr: priority=0 tries=15 successful=0 
     2478088     8388608       5  Label: "ROOT-B"
                                  Type: ChromeOS rootfs
                                  UUID: 716454E2-8860-CB48-BC66-EF2BB23C1CC4
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: F87C9678-7370-A34C-AB6D-6A7FAFB6A180
                                  Attr: priority=0 tries=15 successful=0 
       16456     2097152       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: FF2C1097-D572-1747-8409-5960D20DEC8C
     2248712       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: 03BC5A37-A26B-2844-88C4-4642192623F2
     2113608           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 775282D3-6533-7D41-BD66-26F45F366F98
     2113609           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: FB2F3A35-5E22-444E-A567-A7BD6164F55D
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: E683B39F-806F-4F4B-A29F-FFCB9A02603B
     2412552       65536      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: 7673DF71-A813-4643-92AB-2DDB6BC5AD65
                                  Attr: legacy_boot=1 
    52428767          32          Sec GPT table
    52428799           1          Sec GPT header
Writing partition 1
8,95MiB 0:00:00 [55,4MiB/s] [<=>                                               ]
Writing partition 2
16,0MiB 0:00:00 [57,4MiB/s] [<=>                                               ]
Writing partition 3
2,00GiB 0:01:26 [23,7MiB/s] [              <=>                                 ]
Writing partition 4
16,0MiB 0:00:00 [45,6MiB/s] [<=>                                               ]
Writing partition 5
2,00GiB 0:01:33 [21,8MiB/s] [      <=>                                         ]
Writing partition 6
 512 B 0:00:00 [9,21MiB/s] [<=>                                                ]
Writing partition 7
1,00GiB 0:00:37 [27,2MiB/s] [                                   <=>            ]
Writing partition 8
16,0MiB 0:00:00 [48,3MiB/s] [<=>                                               ]
Writing partition 9
 512 B 0:00:00 [ 361KiB/s] [<=>                                                ]
Writing partition 10
 512 B 0:00:00 [1,61MiB/s] [<=>                                                ]
Writing partition 11
8,00MiB 0:00:00 [48,1MiB/s] [<=>                                               ]
Writing partition 12
32,0MiB 0:00:00 [48,6MiB/s] [<=>                                               ]
ChromeOS installed.
`
p/s: My english was pretty suck, sorry for that...

sebanc, 2020-12-06:
Most reports say that dual boot won"t work with ubuntu mbr grub but someone told me he had succeeded with arch grub.

As long as you know how to populate img_path and img_path variables, you can use the below template:
```
menuentry "Brunch" {
    rmmod tpm
    img_part=/dev/xxxxx
    img_path=/chromeos.img
    search --no-floppy --set=root --file $img_path
    loopback loop $img_path
    linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
        cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path \
        console= vt.global_cursor_default=0 brunch_bootsplash=default
    initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
```

glowingnec, 2020-12-07:
> Most reports say that dual boot won"t work with ubuntu mbr grub but someone told me he had succeeded with arch grub.
>
> As long as you know how to populate img_path and img_path variables, you can use [...]

Thank, I will try it, but...what is the different between Ubuntu Grub and Arch Grub?

glowingnec, 2020-12-07:
Well, I have an Arch install running XFCE, but I booting it up using Ubuntu Grub
So, I need to go to Arch, type "grub-install /dev/sda", then add ChromeOS entry, right?

glowingnec, 2020-12-08:
It's now only showing "no such module" when boot, press any key will boot the kernel....
And it hang there...[image]