sebanc/brunch #687: MacBook Pro Always Auto reboot

MrHongping, 2020-12-06:
md101 cpu is i5 3210m Graphics hd 4000 I tried several versions with the same result 
[image]
[image]

sebanc, 2020-12-06:
The error here is not hardware related but due to an incorrect install. Try again the install after re-downloading the "samus" recovery image and latest brunch.

MrHongping, 2020-12-06:
> The error here is not hardware related but due to an incorrect install. Try again the install after re-downloading the "samus" recovery image and latest brunch.

Ok, thank you for your answer. I'll keep trying.

MrHongping, 2020-12-07:
[image][image]

I rebuilt the startup program and got such an outcome.

closed 2026-03-31 by github-actions[bot]