sebanc/brunch #689: New ipts driver's issue when connecting to external display

bbs3223474, 2020-12-06:
My device: Surface Book 2 Core-i5/8GB w/o DGPU
ChromeOS version: 86
Brunch version K5.4

At this brunch release, my Surface book2's touchscreen will shift left when connecting to an external display (monitor, capture card, etc) and using mirror mode. For example, if I touch "minimize" button on the Chrome window, it will actually hit the "close" button. If I want to click notification bar to change the brightness, I have to touch 2-3cm left to actually hit it.
Disconnecting the external display device will put everything back to normal.
In this case, I'm using a capture card that's able to process 1080P 60FPS signal. I can't be sure if that will affect the touchscreen or not since Surface book uses 3000x2000 resolution.
Previous brunch kernel doesn't have this issue.

agentICS, 2020-12-07:
(Moved from issue #690)

Hi there, I recently updated Brunch to the lastest version. My Surface Pro 4 even with Intel-ipts.no-feedback=1 sometimes lost touch sometimes however with this new IPTS driver the touchscreen seldomly works now.

The touchscreen jumps whenever you long press it on the sides I've noticed, I thought it was my Surface Pro's hardware however this does not affect Windows. Taps are also not detected all the time by the touchscreen, sometimes Chrome OS doesn't detected taps on the sides, I can try to long press and Chrome OS thinks I tapped once, I can drag my finger across in a drawing app only to find that the line is patchy. This has left the touchscreen basically unusable.

Is there possibly any remedy to this? Thank you

Note: Just to clarify, I did add the options=ipts option after cros_debug in GRUB.

sebanc, 2020-12-07:
Hi,

First, a little history on the ipts driver:
Up until kernel 5.2, brunch was using an unofficial ipts driver created by intel guys here https://github.com/ipts-linux-org/ipts-linux-new/wiki and that was maintained by the community (jakeday / Linux-surface team). This driver worked with something called GuC in the intel i915 kernel driver.
As of kernel 5.3, Intel removed the GuC functionality in its kernel driver and we were left without solution for the ipts driver.
The linux-surface team (iirc mostly StollD and qzed) did a huge work to find a solution to this situation and managed to create a new ipts driver that does not use the GuC. This driver still has a few bugs but is being actively worked on and is the only existing solution for kernel versions above 5.3.

Wrt brunch, I am currently trying to figure out the best way to maintain both the 4.19 kernel and the 5.4 kernel so that IPTS users can continue using the 4.19 kernel until the main issues in the new ipts driver are fixed. Nevertheless, it is not really a long term solution...

For now, if you have issues with the ipts driver I suggest staying on the latest 4.19 kernel build. I will try to provide an updated brunch 4.19 version soon.

agentICS, 2020-12-07:
> Hi,
> 
> First, a little history on the ipts driver:
> Up until kernel 5.2, the ipts driver used in Surface devices compatible linux kernel was an unofficial ipts driver created by intel guys here https://github.com/ipts-linux-org/ipts-linux-new/wiki that was maintained by the community (jakeday / Linux-surface team). This driver worked with something called GuC in the intel i915 kernel driver.
> As of kernel 5.3, Intel removed the GuC functionality in its kernel driver and we were left without solution for the ipts driver.
> The linux-surface team (iirc mostly StollD and qzed) did a huge work to find a solution to this situation and managed to create a new ipts driver that does not use the GuC. This driver still has a few bugs but is being actively worked on and is the only existing solution for kernel versions above 5.3.
> 
> Wrt brunch, I am currently trying to figure out the best way to maintain both the 4.19 kernel and the 5.4 kernel so that IPTS users can continue using the 4.19 kernel until the main issues in the new ipts driver are fixed. Nevertheless, it is not really a long term solution...
> 
> For now, if you have issues with the ipts driver I suggest staying on the latest 4.19 kernel build. I will try to provide an updated brunch 4.19 version soon.

Thanks for the history behind the IPTS driver! Would it be possible to "downgrade" the brunch framework (not Chrome OS) or would a full reinstall be needed? Thanks

sebanc, 2020-12-07:
You should be able to downgrade but some who previously tried this have apparently lost their data in the process, so make a backup of your data first.

santhoshr, 2020-12-10:
Thank you @sebanc for providing history on the issue, I too updated on by SB2 and found touch to be jittery in new 5.x kernel release, it would be cool if you can maintain two updated release for 4.x and 5.x kernel lines. The 4.x version is much much better even comparing up to date Windows 10 as well, most likely the intel GuC affected Windows drivers as well. I love the new brunch boot screen so eagerly waiting for up-to-date 4.x brunch release.

On unrelated few things, 
i) any way to hide blindingly bright Chrome boot screen, even with brunch boot screen, the end has the bright Chrome screen ?
ii) Can chrome os detect and display status of two surface book 2 batteries ?
iii) any tool or ways to restrict CPU throttling like TLP service, https://linrunner.de/tlp/, sometimes the Clipboard (tablet part of SB2) heats up
iv) thanks to your boot option I was able to force tablet mode using alias or small script, any way to create shortcut in the launcher for the same ? or option in notification tray, place where wifi, bluetooth options are ?

sebanc, 2020-12-11:
brunch-testing repository is now public and contains an unstable release with both 5.4 and 4.19 kernel (5.4 is the default, if you want to boot with 4.19 kernel replace "/kernel" on the grub configuration linux line with "/kernel-4.19").

@santhoshr
1) Wisteria is working on a method to change the bootanimation but it is not completely finalised yet:
https://github.com/WesBosch/brunch-toolkit
2) Chromeos only supports an aggregated battery status when there are several batteries
3) I don't think tlp in general is a good solution so I am not willing to include it but you can probably build tlp with the "brunch-toolchain":
https://github.com/sebanc/brunch-toolchain
4) No, that would require modifying the chromeos source code which is not possible with brunch

rexsububi, 2021-02-19:
Just a heads up for anyone following this thread on resolving touchscreen issues on a surface 3 tablet; I ran into issue #180 since Intel Iris Plus Graphics are not properly supported by 4.19. Best support I have had so far is editing to "/kernel-5.10", which resolves camera support but I still do not have touchscreen support.