sebanc/brunch #690: Screen jumping and non detection of fingers with new IPTS driver (Surface Pro 4)

agentICS, 2020-12-07:
Hi there, I recently updated Brunch to the lastest version. My Surface Pro 4 even with Intel-ipts.no-feedback=1 sometimes lost touch sometimes however with this new IPTS driver the touchscreen seldomly works now.

The touchscreen jumps whenever you long press it on the sides I've noticed, I thought it was my Surface Pro's hardware however this does not affect Windows. Taps are also not detected all the time by the touchscreen, sometimes Chrome OS doesn't detected taps on the sides, I can try to long press and Chrome OS thinks I tapped once, I can drag my finger across in a drawing app only to find that the line is patchy. This has left the touchscreen basically unusable.

Is there possibly any remedy to this? Thank you

Note: Just to clarify, I did add the options=ipts option after cros_debug in GRUB.

sebanc, 2020-12-07:
I will try to maintain all the new IPTS driver issues in #689, could you copy your issue there in a comment ?

agentICS, 2020-12-07:
Just did, I'll close this issue.

closed 2020-12-07 by agentICS