sebanc/brunch #691: Camera not working with latest samus v87

Crescendo-BLYAT, 2020-12-08:
`chronos@localhost / $ dmesg | grep camera
[   71.503283] brunch: /firmware/patches/99-camera.sh success
[   96.547815] init: cros-camera main process ended, respawning
[  102.006537] init: cros-camera main process ended, respawning
[  107.332544] init: cros-camera main process ended, respawning
[  112.547764] init: cros-camera main process ended, respawning
[  117.726578] init: cros-camera main process ended, respawning
[  122.931615] init: cros-camera main process ended, respawning
[  128.097966] init: cros-camera main process ended, respawning
[  133.272549] init: cros-camera main process ended, respawning
[  138.450578] init: cros-camera main process ended, respawning
[  143.626321] init: cros-camera main process ended, respawning
[  148.827752] init: cros-camera respawning too fast, stopped`

Any app that require camera such as Microsoft Teams will just freeze on opening.

Brunch ver: Brunch r86 k5.4 stable 20201201 & Brunch r87 k5.4 unstable 20201205.

Crescendo-BLYAT, 2020-12-08:
More info using dmesg:

[    4.569840] Run /init as init process
[    4.615526] haveged: command socket is listening at fd 3
[    4.616881] usb 1-5: New USB device found, idVendor=8087, idProduct=0aa7, bcdDevice= 0.01
[    4.616888] usb 1-5: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    4.735780] usb 1-7: new high-speed USB device number 4 using xhci_hcd
[    4.880121] usb 1-7: New USB device found, idVendor=0408, idProduct=a010, bcdDevice= 0.17
[    4.880124] usb 1-7: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[    4.880126] usb 1-7: Product: VGA Webcam
[    4.880128] usb 1-7: Manufacturer: Quanta
[    4.880130] usb 1-7: SerialNumber: NULL
[    4.882692] uvcvideo: Found UVC 1.00 device VGA Webcam (0408:a010)
[    4.882910] uvcvideo: Failed to query (GET_INFO) UVC control 3 on unit 1: -32 (exp. 1).
[    4.884614] uvcvideo: Failed to query (GET_INFO) UVC control 12 on unit 2: -32 (exp. 1).
[    4.891392] input: VGA Webcam: VGA Webcam as /devices/pci0000:00/0000:00:15.0/usb1/1-7/1-7:1.0/input/input10
[    4.922865] haveged: ver: 1.9.13; arch: x86; vend: ; build: (gcc 10.2.0 T); collect: 128K
[    4.922878] haveged: cpu: (); data: 16K (D); inst: 16K (D); idx: 13/40; sz: 15962/54019
[    4.923231] haveged: tot tests(BA8): A:1/1 B:1/1 continuous tests(B):  last entropy estimate 7.999
[    4.923242] haveged: fills: 0, generated: 0

sebanc, 2020-12-08:
Did you try 5.4 kernel previously or were those releases the first 5.4 kernel ones you tried ?

Crescendo-BLYAT, 2020-12-09:
I only tried the Brunch r86 k5.4 stable 20201201 along with samus v87 (OTA update).
Then the camera stopped working after reboot.
Then I tried this Brunch r87 k5.4 unstable 20201205, reboot: same result.

Only the testing one is faster coz the Brunch r86 k5.4 stable 20201201 got the CPU pegged to 100% almost all the time even after fresh boot & nothing's running.
All laggy even the mouse cursor movements.

sebanc, 2020-12-09:
Thanks ! That's very helpful :)

For the CPU, it seems there is a bug in the most recent chromiumos kernel (I came back to an earlier kernel version between Brunch r86 k5.4 stable 20201201 and Brunch r87 k5.4 unstable 20201205).

For the camera, there are issues reported on chromebooks with r87. Could you try going back to samus r86 and see if you have the same camera issue ? It will make it clearer whether the camera issue is related to chromeos or to the 5.4 kernel.
(back up your data before doing this as in some cases downgrading chromeos prevented users from accessing their data)

Crescendo-BLYAT, 2020-12-09:
You're welcome.
I'll test it tomorrow as I need to make the USB for it in office. 
I don't have computer at home other than Chromebook (this Brunchbook & real Chromebook). LOL.

Crescendo-BLYAT, 2020-12-10:
@sebanc same dmesg uvcvideo error but:

chronos@localhost / $ dmesg | grep camera
[ 71.503283] brunch: /firmware/patches/99-camera.sh success
[ 96.547815] init: cros-camera main process ended, respawning

& THE CAMERA IS WORKING!
samus v86 - Brunch r86 k5.4 stable 20201201

Crescendo-BLYAT, 2020-12-10:
Plus on start of crosh, there's an error:

[ERROR:src/main.rs:248] Error loading history: No such file or directory (os error 2)

Crescendo-BLYAT, 2020-12-11:
Brunch r87 k5.4 unstable 20201205 - samus v87 - WORKING PERFECTLY!!
Now switching to octopus v87 due to AUE to samus - WORKING PERFECTLY!!

closed 2020-12-11 by Crescendo-BLYAT