sebanc/brunch #694: Touchscreen stops working immediately after logging in

bdodroid, 2020-12-08:
On my Surface Go 2 the touchscreen works perfectly fine (including pen) until I login, at which point touch/pen becomes completely unresponsive.

I'm running r86 k5.4 stable 20201201 and have added the ipts option to the kernel line.

Am I missing something? or is the ipts not compatible with surface go 2?

sebanc, 2020-12-08:
Surafec Go devices are not concerned by ipts, try removing this option.

bdodroid, 2020-12-08:
I have tried with and without ipts option and I get the exact same result unfortunately

bdodroid, 2020-12-08:
I did a powerwash and so far everything seems to be working now. 

I'm not sure what exactly I did that would have cause touch/pen to fail only after logging in but I must have done something it didn't like. 

Things I did:
added iwlwifi_backport to grub
enabled Hyper threading in the chrome flags 
enabled Linux beta
Installed Crossover for Chrome OS in crostini 
Installed Shadow via android app store

closed 2020-12-09 by bdodroid