sebanc/brunch #701: Google Play not installing

riqueps, 2020-12-10:
Hello, I've installed Brunch r86 k5.4 stable 20201201 on my acer laptop, but google play are not working.
It try to install but after some minutes it shows an error, "Something went wrong". 
Could you help me?

sebanc, 2020-12-11:
Hi, sorry but unfortunately some 10th gen devices are not yet compatible with the android container and I don't think I can do something about this.

sebanc, 2020-12-11:
You could try to reset Chrome sync but I don't think it will help in this case:

Login to your Google dashboard account -> Go to the "Chrome" application -> Manage synced data and click on "reinitialize synced data"

riqueps, 2020-12-11:
> You could try to reset Chrome sync but I don't think it will help in this case:
> 
> Login to your Google dashboard account -> Go to the "Chrome" application -> Manage synced data and click on "reinitialize synced data"

okay, I understand.
Could you tell me if this issue will be fix in the future?
Thanks in advanced

riqueps, 2020-12-11:
> Hi, sorry but unfortunately some 10th gen devices are not yet compatible with the android container and I don't think I can do something about this.

okay, I understand.
Could you tell me if this issue will be fix in the future?
Thanks in advanced

riqueps, 2020-12-11:
> You could try to reset Chrome sync but I don't think it will help in this case:
> 
> Login to your Google dashboard account -> Go to the "Chrome" application -> Manage synced data and click on "reinitialize synced data"

I've already tried do this, but didn't work.

sebanc, 2020-12-12:
Google is apparently planning to switch to ARCVM which will change a lot of things wrt the android framework and could potentially fix this issue. However, AFAIK there is no official ETA for this change.

princepjamie06832, 2021-03-27:
#983

closed 2026-03-31 by github-actions[bot]